gnif/vendor-reset #3: Not working for 5700XT under 5.8.0

johncolby, 2020-11-19:
Hi guys, thanks for the work, I know it's super early...

I hope this info is useful. Let me know if there is anything else I can add.

## System info

```
❯ uname -a
Linux kvm-host 5.8.0-28-generic #30 SMP Wed Nov 18 22:03:39 PST 2020 x86_64 x86_64 x86_64 GNU/Linux

❯ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:    20.04
Codename:   focal

❯ lsmod | grep vendor
vendor_reset           90112  0

❯ virsh version --daemon
Compiled against library: libvirt 6.0.0
Using library: libvirt 6.0.0
Using API: QEMU 6.0.0
Running hypervisor: QEMU 4.2.1
Running against daemon: 6.0.0

❯ lspci -nnv | grep -A 10 VGA
0a:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] [1002:731f] (rev c1) (prog-if 00 [VGA controller])
	Subsystem: Sapphire Technology Limited Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] [1da2:e411]
	Flags: bus master, fast devsel, latency 0, IRQ 40
	Memory at 7fe0000000 (64-bit, prefetchable) [size=256M]
	Memory at 7ff0000000 (64-bit, prefetchable) [size=2M]
	I/O ports at e000 [size=256]
	Memory at fcd00000 (32-bit, non-prefetchable) [size=512K]
	Expansion ROM at fcd80000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: vfio-pci
	Kernel modules: amdgpu
```

## Logs

First 2 successful resets are upon initial successful boot of win10 vm. Third unsuccessful reset (screen stays black) is when attempting a win10 soft restart.

```
❯ sudo dmesg | grep NAVI
[  316.161395] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[  316.161395] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[  316.161483] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[  316.493688] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[  316.493693] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  316.493695] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[  316.513247] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[  316.825275] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[  316.825277] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[  316.825378] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[  317.183124] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[  317.183129] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  317.183132] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[  317.205125] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[  570.655331] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[  570.655332] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[  570.655420] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[  570.759720] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[  570.759725] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 1e77c78a, mp1 intr enabled? yes, bl ready? no
[  570.759725] vfio-pci 0000:0a:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  570.759850] vfio-pci 0000:0a:00.0: AMD_NAVI10: begin psp mode 1 reset
[  571.269270] vfio-pci 0000:0a:00.0: AMD_NAVI10: mode1 reset succeeded
[  582.072867] vfio-pci 0000:0a:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset
[  582.072874] vfio-pci 0000:0a:00.0: AMD_NAVI10: failed to reset device
[  582.072875] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[  582.072992] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
```

FWIW, resetting works fine under 5.8.0 with the prior v2 reset patch:

```
❯ sudo dmesg | grep -B 3 -A 3 -m 1 Navi
[  844.986707] br0: port 2(vnet0) entered disabled state
[  845.237468] usb 1-6: reset full-speed USB device number 2 using xhci_hcd
[  845.520840] hid-generic 0003:1B1C:0C20.000E: hiddev0,hidraw0: USB HID v1.11 Device [Corsair Inc. H100iRGBPROXT] on usb-0000:02:00.0-6/input0
[  845.632757] vfio-pci 0000:0a:00.0: Navi10: performing BACO reset
[  847.157571] audit: type=1400 audit(1605774819.673:59): apparmor="STATUS" operation="profile_remove" profile="unconfined" name="libvirt-9d1baedf-170f-4646-b760-4e485f2ec1fb" pid=3687 comm="apparmor_parser"
[  847.188645] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC1220: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:line
[  847.188648] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
```

Equivocal90, 2020-11-20:
I'm also having issues with this module on a 5700 XT except under kernel 5.9
The actual GPU works for 1 reset but the Audio part fails to start back up. When the GPU starts failing to reset and I get only a black screen I'm getting the below lines in my `dmesg` output.
`[43469.858696] vendor-reset-drm: invalid ip discovery binary signature
[43469.858697] vendor-reset-drm: amdgpu_discovery_init failed
`

giganode, 2020-11-20:
> I'm also having issues with this module on a 5700 XT except under kernel 5.9
> The actual GPU works for 1 reset but the Audio part fails to start back up. When the GPU starts failing to reset and I [...]

 @Equivocal90 Are you also using a non reference card? Maybe even specifically a Sapphire, too? Like @johncolby ?

Equivocal90, 2020-11-20:
@giganode I am also using a non-reference card. Except mine is a Gigabyte.

```
20:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] [1002:731f] (rev c1) (prog-if 00 [VGA controller])
        Subsystem: Gigabyte Technology Co., Ltd Radeon RX 5700 XT Gaming OC [1458:2313]
        Flags: bus master, fast devsel, latency 0, IRQ 100
        Memory at c0000000 (64-bit, prefetchable) [size=256M]
        Memory at d0000000 (64-bit, prefetchable) [size=2M]
        I/O ports at d000 [size=256]
        Memory at fd300000 (32-bit, non-prefetchable) [size=512K]
        Expansion ROM at fd380000 [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: vfio-pci
        Kernel modules: amdgpu
```

gmt, 2020-11-21:
I share this exact disease with the PowerColor Liquid Devil.

gnif, 2020-11-21:
Please be aware our focus has been on Polaris and Vega 10 currently, I will be putting some work into Navi 10 hopefully some time during the coming month.

gmt, 2020-11-22:
Hm,....  probably OT at this point, but based on this, I tried reassigning vfio to my vega10 and amdgpu to my navi10 card, and managed to pt my vega10 gpu.

So this does indeed seem to work better (I'm able to warm reboot the guest!).  But, once I shut the guest down, I get this mess:

```[21543.091762] vfio-pci 0000:45:00.0: AMD_VEGA10: version 1.0
[21543.091764] vfio-pci 0000:45:00.0: AMD_VEGA10: performing pre-reset
[21543.091830] vfio-pci 0000:45:00.0: AMD_VEGA10: performing reset
[21543.402516] ATOM BIOS: 113-D0501100-109
[21543.402518] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[21543.402521] vfio-pci 0000:45:00.0: AMD_VEGA10: bus reset disabled? yes
[21543.402525] vfio-pci 0000:45:00.0: AMD_VEGA10: SMU response reg: 1, sol reg: 6b0, mp1 intr enabled? no, bl ready? no, baco? off
[21543.616629] vfio-pci 0000:45:00.0: SMU error 0x0
[21543.616630] vfio-pci 0000:45:00.0: Failed to send message 0x6: return 0x0
[21543.616631] vfio-pci 0000:45:00.0: AMD_VEGA10: enabled features: bf4d9000
[21543.616632] vfio-pci 0000:45:00.0: AMD_VEGA10: disabling features
[21543.831112] vfio-pci 0000:45:00.0: SMU error 0x0
[21544.045347] vfio-pci 0000:45:00.0: SMU error 0x0
[21544.259680] vfio-pci 0000:45:00.0: SMU error 0x0
[21544.473895] vfio-pci 0000:45:00.0: SMU error 0x0
[21544.687984] vfio-pci 0000:45:00.0: SMU error 0x0
[21544.687985] vfio-pci 0000:45:00.0: AMD_VEGA10: Driver reset
[21544.902186] vfio-pci 0000:45:00.0: SMU error 0x0
[21544.902188] vfio-pci 0000:45:00.0: AMD_VEGA10: Could not reset w/ PPSMC_MSG_GfxDeviceDriverReset: -110
[21545.409169] vfio-pci 0000:45:00.0: AMD_VEGA10: entering BACO
[21545.623927] vfio-pci 0000:45:00.0: SMU error 0x0
[21545.623930] vfio-pci 0000:45:00.0: AMD_VEGA10: failed to reset device
[21545.623931] vfio-pci 0000:45:00.0: AMD_VEGA10: performing post-reset
[21545.624005] vfio-pci 0000:45:00.0: AMD_VEGA10: reset result = 0
[21582.017514] vfio-pci 0000:45:00.0: enabling device (0400 -> 0403)
[21582.017620] vfio-pci 0000:45:00.0: AMD_VEGA10: version 1.0
[21582.017621] vfio-pci 0000:45:00.0: AMD_VEGA10: performing pre-reset
[21582.017694] vfio-pci 0000:45:00.0: AMD_VEGA10: performing reset
[21582.328199] ATOM BIOS: 113-D0501100-109
[21582.328200] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[21582.328202] vfio-pci 0000:45:00.0: AMD_VEGA10: bus reset disabled? yes
[21582.328207] vfio-pci 0000:45:00.0: AMD_VEGA10: SMU response reg: 0, sol reg: 834, mp1 intr enabled? no, bl ready? no, baco? off
[21582.538586] vfio-pci 0000:45:00.0: SMU error 0x0
[21582.538587] vfio-pci 0000:45:00.0: AMD_VEGA10: could not get enabled SMU features, trying BACO reset anyway [ret -110]
[21582.538588] vfio-pci 0000:45:00.0: AMD_VEGA10: entering BACO
[21582.749611] vfio-pci 0000:45:00.0: SMU error 0x0
[21582.749612] vfio-pci 0000:45:00.0: AMD_VEGA10: failed to reset device
[21582.749613] vfio-pci 0000:45:00.0: AMD_VEGA10: performing post-reset
[21582.749677] vfio-pci 0000:45:00.0: AMD_VEGA10: reset result = 0
[21582.749772] vfio-pci 0000:45:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
[21582.749781] vfio-pci 0000:45:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
[21582.808624] vfio-pci 0000:45:00.0: AMD_VEGA10: version 1.0
[21582.808627] vfio-pci 0000:45:00.0: AMD_VEGA10: performing pre-reset
[21582.808714] vfio-pci 0000:45:00.0: AMD_VEGA10: performing reset
[21583.119727] ATOM BIOS: 113-D0501100-109
[21583.119728] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[21583.119731] vfio-pci 0000:45:00.0: AMD_VEGA10: bus reset disabled? yes
[21583.119735] vfio-pci 0000:45:00.0: AMD_VEGA10: SMU response reg: 0, sol reg: 83c, mp1 intr enabled? no, bl ready? no, baco? off
[21583.334035] vfio-pci 0000:45:00.0: SMU error 0x0
[21583.334036] vfio-pci 0000:45:00.0: AMD_VEGA10: could not get enabled SMU features, trying BACO reset anyway [ret -110]
[21583.334037] vfio-pci 0000:45:00.0: AMD_VEGA10: entering BACO
[21583.548169] vfio-pci 0000:45:00.0: SMU error 0x0
[21583.548170] vfio-pci 0000:45:00.0: AMD_VEGA10: failed to reset device
[21583.548171] vfio-pci 0000:45:00.0: AMD_VEGA10: performing post-reset
[21583.548238] vfio-pci 0000:45:00.0: AMD_VEGA10: reset result = 0
[21590.118810] usb 7-1.4.1: reset low-speed USB device number 5 using xhci_hcd
[21590.758794] usb 7-1.4.3.3: reset low-speed USB device number 8 using xhci_hcd
[21591.142835] usb 7-1.4.3.4: reset full-speed USB device number 9 using xhci_hcd
```

which, although clearly different, ends up pretty much the same (vfio gpu no longer posts, and tear-soaked office chair *sniff*).

I still don't have [c5cb424fe(https://github.com/gnif/vendor-reset/commit/c5cb424fec3d9d27b77367f582f1ac94d892722e)] in my .ko.  I'll pull and rebuild, maybe that's it.  Even if not, looks like a completely different bug...?

giganode, 2020-11-26:
Experiencing identical behaviour on Unraid 6.9.0-beta35 (kernel 5.8.18).
While the 5700XT with reference design works flawlessly, I do have the same issue with the sound card. 
Device properties show a code 10 error.
I did restarts, shutdowns and force stop.

Problem occurred for the first time after windows 10 turned screen off at login because of missing input.

mdjabirov, 2020-11-27:
Can confirm that a reference 5700XT (by Asus) resets fine in both Windows 10 and Linux VMs, but DisplayPort sound would not work after a reset. This is nonetheless an amazing effort and progress!

baderas, 2020-12-06:
Can confirm that is does **not** work for my XFX 5700XT on 5.9.11. After a VM shutdown and boot, the screen stays black and the following dmesg error occurs:

```
[  179.163342] vfio-pci 0000:28:00.1: refused to change power state from D0 to D3hot
[  179.163437] vfio-pci 0000:28:00.0: AMD_NAVI10: version 1.1
[  179.163438] vfio-pci 0000:28:00.0: AMD_NAVI10: performing pre-reset
[  179.163522] vfio-pci 0000:28:00.0: AMD_NAVI10: performing reset
**[  179.164481] vendor-reset-drm: invalid ip discovery binary signature
[  179.164482] vendor-reset-drm: amdgpu_discovery_init failed**
[  179.164483] vfio-pci 0000:28:00.0: AMD_NAVI10: amdgpu_discovery_reg_base_init failed, using legacy method
[  179.268728] ATOM BIOS: 113-230LNAVIXT612_8GD6_MS_W8
[  179.268729] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  179.500492] vfio-pci 0000:28:00.0: AMD_NAVI10: bus reset disabled? yes
[  179.500497] vfio-pci 0000:28:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? no
[  179.500497] vfio-pci 0000:28:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  179.500498] vfio-pci 0000:28:00.0: AMD_NAVI10: begin psp mode 1 reset
[  179.732238] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[  180.491442] vfio-pci 0000:28:00.0: AMD_NAVI10: PSP did not acknowledger reset
**[  180.491537] vfio-pci 0000:28:00.0: AMD_NAVI10: failed to reset device**
[  180.491538] vfio-pci 0000:28:00.0: AMD_NAVI10: performing post-reset
[  180.491638] vfio-pci 0000:28:00.0: AMD_NAVI10: reset result = 0
[  180.506672] vfio-pci 0000:28:00.0: refused to change power state from D0 to D3hot
```

is adding the module to cat /etc/modules-load.d/vendor-reset.conf on archlinux not enough? It seems that vfio-pci is somehow involved, must the module be loaded another way?

Any ideas?

baderas, 2020-12-06:
After writing the comment above, I got the idea to load it before vfio modules by loading it in /etc/mkinitcpio.conf Modules hook (where the vfio modules are also loaded).

It then 'kind of works', but after the second start I only got 800x600 with MS Basic driver or something like that. Dmesg output:

```
[  112.586721] vfio-pci 0000:28:00.0: enabling device (0400 -> 0403)
[  112.586859] vfio-pci 0000:28:00.0: AMD_NAVI10: version 1.1
[  112.586859] vfio-pci 0000:28:00.0: AMD_NAVI10: performing pre-reset
[  112.586951] vfio-pci 0000:28:00.0: AMD_NAVI10: performing reset
[  112.692656] ATOM BIOS: 113-230LNAVIXT612_8GD6_MS_W8
[  112.692658] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  112.692660] vfio-pci 0000:28:00.0: AMD_NAVI10: bus reset disabled? yes
[  112.692665] vfio-pci 0000:28:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 7f7bb27, mp1 intr enabled? yes, bl ready? no
[  112.692665] vfio-pci 0000:28:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  112.692794] vfio-pci 0000:28:00.0: AMD_NAVI10: begin psp mode 1 reset
[  112.925435] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[  113.436728] vfio-pci 0000:28:00.0: AMD_NAVI10: mode1 reset succeeded
[  124.103394] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset
[  124.103398] vfio-pci 0000:28:00.0: AMD_NAVI10: failed to reset device
[  124.103399] vfio-pci 0000:28:00.0: AMD_NAVI10: performing post-reset
[  124.103514] vfio-pci 0000:28:00.0: AMD_NAVI10: reset result = 0
[  124.103663] vfio-pci 0000:28:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
[  124.103675] vfio-pci 0000:28:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
[  124.103680] vfio-pci 0000:28:00.0: vfio_ecap_init: hiding ecap 0x25@0x400
[  124.103682] vfio-pci 0000:28:00.0: vfio_ecap_init: hiding ecap 0x26@0x410
[  124.103684] vfio-pci 0000:28:00.0: vfio_ecap_init: hiding ecap 0x27@0x440
[  124.203494] vfio-pci 0000:28:00.0: AMD_NAVI10: version 1.1
[  124.203496] vfio-pci 0000:28:00.0: AMD_NAVI10: performing pre-reset
[  124.203587] vfio-pci 0000:28:00.0: AMD_NAVI10: performing reset
[  124.309536] ATOM BIOS: 113-230LNAVIXT612_8GD6_MS_W8
[  124.309537] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  124.309539] vfio-pci 0000:28:00.0: AMD_NAVI10: bus reset disabled? yes
[  124.309544] vfio-pci 0000:28:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 7f7bb27, mp1 intr enabled? yes, bl ready? no
[  124.309544] vfio-pci 0000:28:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  124.309668] vfio-pci 0000:28:00.0: AMD_NAVI10: begin psp mode 1 reset
[  124.542240] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[  125.063378] vfio-pci 0000:28:00.0: AMD_NAVI10: mode1 reset succeeded
[  135.760046] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset
[  135.760050] vfio-pci 0000:28:00.0: AMD_NAVI10: failed to reset device
[  135.760051] vfio-pci 0000:28:00.0: AMD_NAVI10: performing post-reset
[  135.760167] vfio-pci 0000:28:00.0: AMD_NAVI10: reset result = 0
[  206.667656] vfio-pci 0000:28:00.0: AMD_NAVI10: version 1.1
[  206.667658] vfio-pci 0000:28:00.0: AMD_NAVI10: performing pre-reset
[  206.667747] vfio-pci 0000:28:00.0: AMD_NAVI10: performing reset
[  206.776281] ATOM BIOS: 113-230LNAVIXT612_8GD6_MS_W8
[  206.776282] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  206.776284] vfio-pci 0000:28:00.0: AMD_NAVI10: bus reset disabled? yes
[  206.776289] vfio-pci 0000:28:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 7f7bb27, mp1 intr enabled? yes, bl ready? no
[  206.776289] vfio-pci 0000:28:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  206.776337] vfio-pci 0000:28:00.0: AMD_NAVI10: begin psp mode 1 reset
[  207.009204] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[  207.516669] vfio-pci 0000:28:00.0: AMD_NAVI10: mode1 reset succeeded
[  218.186660] vfio-pci 0000:28:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset
[  218.186666] vfio-pci 0000:28:00.0: AMD_NAVI10: failed to reset device
[  218.186667] vfio-pci 0000:28:00.0: AMD_NAVI10: performing post-reset
[  218.186785] vfio-pci 0000:28:00.0: AMD_NAVI10: reset result = 0
```

aliaj00, 2020-12-08:
does not work centos 8 kernel 5.9.13

[root@***** ~]# uname -a
Linux ***** 5.9.13-1.el8.elrepo.x86_64 #1 SMP Mon Dec 7 11:45:32 EST 2020 x86_64 x86_64 x86_64 GNU/Linux
[root@**** ~]# lsmod | grep vendor
vendor_reset          102400  0
[root@**** ~]# 
[root@**** ~]# lspci -nnv | grep -A 10 VGA
29:00.0 VGA compatible controller [0300]: NVIDIA Corporation GK208B [GeForce GT 710] [10de:128b] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: ZOTAC International (MCO) Ltd. Device [19da:5360]
	Flags: bus master, fast devsel, latency 0, IRQ 104
	Memory at f6000000 (32-bit, non-prefetchable) [size=16M]
	Memory at e8000000 (64-bit, prefetchable) [size=128M]
	Memory at f0000000 (64-bit, prefetchable) [size=32M]
	I/O ports at c000 [size=128]
	Expansion ROM at f7000000 [disabled] [size=512K]
	Capabilities: [60] Power Management version 3
	Capabilities: [68] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [78] Express Legacy Endpoint, MSI 00
--
2d:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] [1002:731f] (rev c1) (prog-if 00 [VGA controller])
	Subsystem: Sapphire Technology Limited Radeon RX 5600 XT [1da2:e411]
	Flags: fast devsel, IRQ 40
	Memory at d0000000 (64-bit, prefetchable) [size=256M]
	Memory at e0000000 (64-bit, prefetchable) [size=2M]
	I/O ports at f000 [size=256]
	Memory at f7a00000 (32-bit, non-prefetchable) [size=512K]
	Expansion ROM at f7a80000 [disabled] [size=128K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [64] Express Legacy Endpoint, MSI 00
====================================================================================
io-pci 0000:2d:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  +0.000143] vfio-pci 0000:2d:00.0: AMD_NAVI10: begin psp mode 1 reset
[  +0.508504] vfio-pci 0000:2d:00.0: AMD_NAVI10: mode1 reset succeeded
[ +10.400311] vfio-pci 0000:2d:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset
[  +0.000007] vfio-pci 0000:2d:00.0: AMD_NAVI10: failed to reset device
[  +0.000001] vfio-pci 0000:2d:00.0: AMD_NAVI10: performing post-reset
[  +0.000105] vfio-pci 0000:2d:00.0: AMD_NAVI10: reset result = 0
[  +0.145616] vfio-pci 0000:2d:00.0: AMD_NAVI10: version 1.1
[  +0.000001] vfio-pci 0000:2d:00.0: AMD_NAVI10: performing pre-reset
[  +0.000093] vfio-pci 0000:2d:00.0: AMD_NAVI10: performing reset
[  +0.000949] vendor-reset-drm: invalid ip discovery binary signature
[  +0.000002] vendor-reset-drm: amdgpu_discovery_init failed
[  +0.000001] vfio-pci 0000:2d:00.0: AMD_NAVI10: amdgpu_discovery_reg_base_init failed, using legacy method
[  +0.102784] ATOM BIOS: 113-1E4112U-O45
[  +0.000001] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  +0.230978] vfio-pci 0000:2d:00.0: AMD_NAVI10: bus reset disabled? yes
[  +0.000004] vfio-pci 0000:2d:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? no
[  +0.000001] vfio-pci 0000:2d:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[  +0.000001] vfio-pci 0000:2d:00.0: AMD_NAVI10: begin psp mode 1 reset
[  +0.231163] vfio-pci 0000:2d:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[  +0.732936] vfio-pci 0000:2d:00.0: AMD_NAVI10: PSP did not acknowledger reset
[  +0.000087] vfio-pci 0000:2d:00.0: AMD_NAVI10: failed to reset device
[  +0.000000] vfio-pci 0000:2d:00.0: AMD_NAVI10: performing post-reset
[  +0.000091] vfio-pci 0000:2d:00.0: AMD_NAVI10: reset result = 0
[  +4.193741] vfio-pci 0000:2d:00.1: refused to change power state from D0 to D3hot
[  +3.292977] vfio-pci 0000:2d:00.1: refused to change power state from D0 to D3hot

this frustrating :(

kawogi, 2020-12-16:
Not sure if that belongs here. I had a lot of trouble getting my Sapphire Radeon RX 5700 XT to run with kernel 5.9 (latest Manjaro) in Win10-vm. First boot after power-cycle was always fine. Second boot was random: sometimes fine as well, sometimes black, sometimes low-res. (running a 3d benchmark was a guarantee for not surviving a restart). Sometimes bringing the host into suspend and back made the card work again.
Since I installed the latest Radeon drivers everything seems to work perfect. Didn't test audio so far.

The most prominent error (imo) was `kernel: vfio-pci 0000:10:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset`

Hardware: `IOMMU Group 29:	10:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] [1002:731f] (rev c1)`

I have recorded some successful and failed starts and stops if that helps. But I'm not sure if that is the problem described in here.

gmt, 2020-12-16:
> The most prominent error (imo) was `kernel: vfio-pci 0000:10:00.0: AMD_NAVI10: timed out waiting for PSP bootloader to respond after reset`

That's what a vegetative amd card looks like to vendor-reset.  The lights are on, but nobody's home.

That's generally where you'll find yourself after vendor-reset attempts to work some magic and things go pear-shaped.

The most diagnostically useful stuff will appear in dmesg between the last time it said "reset succeeded" (if ever), and the first time it started saying "reset failed".

Typically, nothing good will happen to your card or your operating system's relationship to it, after that.  Plus something bad might occasionally happen -- you'd might as well reboot at this point, unless you're about to start kgdb

Multiple failure modalities have been discussed above but fwiw in OP's case, he was seeing something like:

> vfio-pci ... timed out waiting for PSP bootloader to respond after reset
> vfio-pci ... failed to reset device

johncolby, 2020-12-29:
So from following along here and elsewhere, it does seem as if some 5700 XT boards have been able to be reset using this method, while some have not. Alternatively, there may be some other confounding factors involved (kernel version?). 

@ajmadsen FWIW, I *AM* still able to recover the GPU in linux by rebinding the `amdgpu` driver on the host, which was the original motivation for this module:

```
...
[41104.312305] amdgpu 0000:0a:00.0: amdgpu: GPU reset begin!
[41104.790336] amdgpu 0000:0a:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
[41104.790415] [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KGQ disable failed
[41105.087340] amdgpu 0000:0a:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
[41105.087412] [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KCQ disable failed
[41105.127576] [drm] free PSP TMR buffer
[41108.274891] amdgpu 0000:0a:00.0: amdgpu: GPU reset succeeded, trying to resume
[41108.275053] [drm] PCIE GART of 512M enabled (table at 0x0000008000000000).
[41108.275168] [drm] VRAM is lost due to GPU reset!
[41108.275473] [drm] PSP is resuming...
[41108.451538] [drm] reserve 0xa00000 from 0x81fe400000 for PSP TMR
[41108.643530] amdgpu 0000:0a:00.0: amdgpu: RAS: optional ras ta ucode is not available
[41108.667531] amdgpu: SMU is resuming...
[41108.669510] amdgpu: SMU is resumed successfully!
[41108.818144] [drm] kiq ring mec 2 pipe 1 q 0
[41108.828175] [drm] VCN decode and encode initialized successfully(under DPG Mode).
[41108.828231] [drm] JPEG decode initialized successfully.
[41108.828248] amdgpu 0000:0a:00.0: amdgpu: ring gfx_0.0.0 uses VM inv eng 0 on hub 0
[41108.828249] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.0.0 uses VM inv eng 1 on hub 0
[41108.828250] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.1.0 uses VM inv eng 4 on hub 0
[41108.828251] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.2.0 uses VM inv eng 5 on hub 0
[41108.828252] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.3.0 uses VM inv eng 6 on hub 0
[41108.828253] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.0.1 uses VM inv eng 7 on hub 0
[41108.828254] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.1.1 uses VM inv eng 8 on hub 0
[41108.828255] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.2.1 uses VM inv eng 9 on hub 0
[41108.828256] amdgpu 0000:0a:00.0: amdgpu: ring comp_1.3.1 uses VM inv eng 10 on hub 0
[41108.828257] amdgpu 0000:0a:00.0: amdgpu: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
[41108.828258] amdgpu 0000:0a:00.0: amdgpu: ring sdma0 uses VM inv eng 12 on hub 0
[41108.828259] amdgpu 0000:0a:00.0: amdgpu: ring sdma1 uses VM inv eng 13 on hub 0
[41108.828260] amdgpu 0000:0a:00.0: amdgpu: ring vcn_dec uses VM inv eng 0 on hub 1
[41108.828261] amdgpu 0000:0a:00.0: amdgpu: ring vcn_enc0 uses VM inv eng 1 on hub 1
[41108.828262] amdgpu 0000:0a:00.0: amdgpu: ring vcn_enc1 uses VM inv eng 4 on hub 1
[41108.828263] amdgpu 0000:0a:00.0: amdgpu: ring jpeg_dec uses VM inv eng 5 on hub 1
[41108.856283] [drm] recover vram bo from shadow start
[41108.856348] [drm] recover vram bo from shadow done
[41108.856361] [drm] Skip scheduling IBs!
[41108.856371] amdgpu 0000:0a:00.0: amdgpu: GPU reset(1) succeeded!
```

ajmadsen, 2020-12-29:
@johncolby was that after a reset with the module failed, or in lieu of the module, but following an event requiring device reset?

johncolby, 2020-12-30:
Hey Adam! I was talking about the latter: In lieu of the module, but following an event requiring device reset. In that baseline case, like your previous experience, I can rebind the `amdgpu` driver and recover normal display output on the host.

I just tried the former, rebinding `amdgpu` after the module failed to reset (after a win10 restart, as in my original post, above), and it did not work:

```
❯ sudo modprobe amdgpu
Killed
```
(no sign of life on the display; host requires a hard restart.)

---
Interestingly, I just saw [this](https://www.reddit.com/r/VFIO/comments/k58grq/amd_reset_bug_kernal_patch_help/gedrj1l/) comment on reddit re: `vendor-reset` working in a macOS VM, so I thought I'd give it a try (hadn't even bothered to try sooner since the old gnif v2 patch, while it worked pretty well on my win10 VM, *never* worked on my macOS VM). And guess what? **IT WORKED!** Boot, restart, shutdown, all with no problem:

<details>

```
[  542.649623] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[  542.649625] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[  542.649720] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[  542.908695] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[  542.908701] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  542.908704] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[  542.925468] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[  543.233546] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[  543.233548] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[  543.233640] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[  543.489753] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[  543.489758] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  543.489761] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[  543.509438] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[  949.438395] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[  949.438397] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[  949.438481] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[  949.694603] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[  949.694609] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  949.694611] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[  949.714088] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[ 1005.033865] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[ 1005.033865] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[ 1005.033952] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[ 1005.264132] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1005.264137] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[ 1005.264138] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[ 1005.281722] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[ 1005.598243] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[ 1005.598245] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[ 1005.598343] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[ 1005.854484] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1005.854490] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[ 1005.854492] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[ 1005.873724] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[ 1030.274972] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[ 1030.274974] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[ 1030.275058] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[ 1030.519636] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1030.519641] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[ 1030.519642] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[ 1030.537563] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[ 2849.400313] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[ 2849.400315] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[ 2849.400425] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[ 2849.402475] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[ 2849.402480] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 3bec91ab, mp1 intr enabled? yes, bl ready? yes
[ 2849.402481] vfio-pci 0000:0a:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[ 2849.402557] vfio-pci 0000:0a:00.0: AMD_NAVI10: begin psp mode 1 reset
[ 2849.930118] vfio-pci 0000:0a:00.0: AMD_NAVI10: mode1 reset succeeded
[ 2851.765667] vfio-pci 0000:0a:00.0: AMD_NAVI10: PSP mode1 reset successful
[ 2851.765674] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing post-reset
[ 2851.790126] vfio-pci 0000:0a:00.0: AMD_NAVI10: reset result = 0
[ 2910.153422] vfio-pci 0000:0a:00.0: AMD_NAVI10: version 1.1
[ 2910.153424] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing pre-reset
[ 2910.153510] vfio-pci 0000:0a:00.0: AMD_NAVI10: performing reset
[ 2910.155072] vfio-pci 0000:0a:00.0: AMD_NAVI10: bus reset disabled? yes
[ 2910.155077] vfio-pci 0000:0a:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: a0ce5acc, mp1 intr enabled? yes, bl ready? yes
[ 2910.155078] vfio-pci 0000:0a:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[... cut ...]
This has me thinking, together with https://github.com/gnif/vendor-reset/issues/3#issuecomment-745744548 above, other people being able to reset Navi10 in general with `vendor-reset`, and the fact that this specific card reset well under win10 previously with the gnif v2 patch, that the failure mode in this particular case may be more to do with the guest OS specifics. First things that come to mind: This is a win10 LTS VM for my job, which, while kept up-to-date, is an old build. When I have a chance, I will try a fresh VM with current win10 pro. Also, FWIW, I never installed the AMD adrenalin suite/bloat, just the stock windows drivers.

ajmadsen, 2020-12-30:
That is an interesting observation @johncolby. I wonder if it has to do with differences in the firmware loaded by the drivers in the different OSes. When you get the chance, could you test with a Linux guest as well?

However, if that were to be the case, I'm not even sure what could be done about it.

johncolby, 2021-01-04:
Great news: 

I tried a fresh install of the AMD drivers (DDU clean and then manual device manager install of AMD driver payload from the adrenalin package) and now `vendor-reset` is working perfectly in the win10 machine. Here is the specific combo of card hardware, vBIOS, and driver version I am running:

[image]

FWIW I also did another fresh install of win10 Pro and Ubuntu linux, like we discussed, and both of those are working fine too. I have tested restarts and shutdowns on all of these VMs, also going back and forth between them, all without rebooting the host. 

In case others are curious, with CSM enabled in host EFI and vBIOS passed as rom file to guest (as described [elsewhere](https://passthroughpo.st/explaining-csm-efifboff-setting-boot-gpu-manually/)), I can also do single GPU passthrough OK for the first time with this AMD card (shutdown the VM, rebind `amdgpu` and start the display manager on the host, stop the display manager and unbind `amdgpu`, start another VM), although this is more tenuous in general and for my needs I prefer to blacklist the driver completely and run headless.

Here are the guest OS versions I have tested, all working:

- Windows 10 Enterprise LTSC version 1607 (build 14393.4104) 
- Windows 10 Pro version 20H2 (build 19042.631)
- Mac OS Catalina 10.15.7
- Mac OS Big Sur 11.1
- Ubuntu 20.04

So as far as I am concerned this issue is closed. I will leave it up to you guys whether you want to leave it open to keep a thread of other 5700 XT issues. Thanks again for all the work!!! 👍

giganode, 2021-01-04:
@johncolby Does audio work on your system after reboot?

Reference design?

mdjabirov, 2021-01-04:
@johncolby - I also tested the reset on Win 10/Ubuntu/Fedora VMs on a Proxmox host (don't expect too many people to have MacOS VM to work with, so leaving that aside). DP/HDMI audio stops working after the 1st reset. While the GPU resetting is a fantastic result, I think we should leave this open unless you observe different results.

p.s.
I recently obtained a NAVI 6000 series card and reset works properly without the kernel module - both video and sound. As these are still to obtain, and prices are way above MSRP, I think people would be very interested in this project still for many months to come.

johncolby, 2021-01-04:
I don't have an HDMI audio capable display to test thoroughly. However, consistent with what you guys are observing, while I can see the HDMI audio PCI device after reset, I do lose the device as an actual output option in macOS. I understand, though, that this is out-of-scope of the vendor-reset project.

I'm on Sapphire Pulse, a non-reference design; see above.

giganode, 2021-01-04:
An update from my point of view:

I updated to Unraid 6.9.0-rc2 with kernel version 5.10.1.

In macOS as long as I just reboot everything is fine. If I shutdown macOS and start the vm again the sound is gone.

With Windows this happens in every way. 1st boot is okay and with the 2nd the sound is gone.

mdjabirov, 2021-01-04:
MacOS is where DP sound would be very important depending on the type of sound device used. Some USB based sound devices common on motherboards these days do not have community MacOS drivers and DP sound is the only way to go.

You'd want sound functionality to work after a reset (not to take way any credit for getting video reset working, which seems challenging enough). Usually you would pass through the entire IOMMU group associated with the GPU (including the sound device) and the fact that sound works with 'properly' resetting cards such as the Radeon 6000 series, means that (ideally) you want it to work with this quirk as well.

giganode, 2021-01-04:
Luckily the reset bug is finally fixed by amd since the rx6000 series...

ajmadsen, 2021-01-07:
Re: audio, it's a known issue for AMD even when handling resets (as far as I can tell in the linux driver), however I have not figured out how to keep the audio device from erroring on a second boot following reset.

In the mean time, you can rebind the audio device to `snd_hda_intel` after shutting down a VM and it will recover the device for you.

alex-caelus, 2021-01-09:
Hi!

I also have problems with 5700XT audio after 2nd boot. I tried @ajmadsen's workaround but It doesn't appear to work.  This is what I tried, specifically (after I shutdown the VM):

    echo -n "0000:04:00.1" > /sys/bus/pci/drivers/vfio-pci/unbind
    echo -n "0000:04:00.1" > /sys/bus/pci/drivers/snd_hda_intel/bind

This is what dmesg says

    [Jan 9 11:22] snd_hda_intel 0000:04:00.1: enabling device (0000 -> 0002)
    [  +0.000121] snd_hda_intel 0000:04:00.1: Handle vga_switcheroo audio client
    [  +0.000096] snd_hda_intel 0000:04:00.1: number of I/O streams is 30, forcing separate stream tags
    [  +0.102080] snd_hda_intel 0000:04:00.1: CORB reset timeout#2, CORBRP = 65535
    [  +0.000751] hdaudio hdaudioC0D0: no AFG or MFG node found
    [  +0.000739] hdaudio hdaudioC0D1: no AFG or MFG node found
    [  +0.000603] hdaudio hdaudioC0D2: no AFG or MFG node found
    [  +0.000603] hdaudio hdaudioC0D3: no AFG or MFG node found
    [  +0.000584] snd_hda_intel 0000:04:00.1: no codecs initialized

And then:

    echo -n "0000:04:00.1" > /sys/bus/pci/drivers/snd_hda_intel/unbind
    echo -n "0000:04:00.1" > /sys/bus/pci/drivers/vfio-pci/bind

And finally I started my VM again.

Like before, the GPU works perfectly fine, but the audio device cannot be started in VM. But I did get (like usual) the following output from KVM:

    kvm: vfio: Cannot reset device 0000:04:00.1, no available reset mechanism.
    kvm: vfio: Cannot reset device 0000:04:00.1, no available reset mechanism.

Dmesg says nothing about it though:
    
    [Jan 9 11:24] vfio-pci 0000:04:00.0: AMD_NAVI10: version 1.1
    [  +0.000003] vfio-pci 0000:04:00.0: AMD_NAVI10: performing pre-reset
    [  +0.000081] vfio-pci 0000:04:00.0: AMD_NAVI10: performing reset
    [  +0.104373] ATOM BIOS: 115-D199PI0-101
    [  +0.000002] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
    [  +0.167021] vfio-pci 0000:04:00.0: AMD_NAVI10: bus reset disabled? yes
    [  +0.000004] vfio-pci 0000:04:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
    [  +0.000004] vfio-pci 0000:04:00.0: AMD_NAVI10: performing post-reset
    [  +0.016468] vfio-pci 0000:04:00.0: AMD_NAVI10: reset result = 0
    [  +1.914105] device tap126i0 entered promiscuous mode
    [  +0.030151] fwbr126i0: port 1(fwln126i0) entered blocking state
    [  +0.000003] fwbr126i0: port 1(fwln126i0) entered disabled state
    [  +0.000092] device fwln126i0 entered promiscuous mode
    [  +0.000066] fwbr126i0: port 1(fwln126i0) entered blocking state
    [  +0.000001] fwbr126i0: port 1(fwln126i0) entered forwarding state
    [  +0.003687] vmbr0: port 2(fwpr126p0) entered blocking state
    [  +0.000002] vmbr0: port 2(fwpr126p0) entered disabled state
    [  +0.000054] device fwpr126p0 entered promiscuous mode
    [  +0.000030] vmbr0: port 2(fwpr126p0) entered blocking state
    [  +0.000001] vmbr0: port 2(fwpr126p0) entered forwarding state
    [  +0.003638] fwbr126i0: port 2(tap126i0) entered blocking state
    [  +0.000001] fwbr126i0: port 2(tap126i0) entered disabled state
    [  +0.000103] fwbr126i0: port 2(tap126i0) entered blocking state
    [  +0.000001] fwbr126i0: port 2(tap126i0) entered forwarding state
    [ +11.160311] vfio-pci 0000:04:00.0: enabling device (0400 -> 0403)
    [  +0.000123] vfio-pci 0000:04:00.0: AMD_NAVI10: version 1.1
    [  +0.000001] vfio-pci 0000:04:00.0: AMD_NAVI10: performing pre-reset
    [  +0.000065] vfio-pci 0000:04:00.0: AMD_NAVI10: performing reset
    [  +0.103231] ATOM BIOS: 115-D199PI0-101
    [  +0.000002] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
    [  +0.165077] vfio-pci 0000:04:00.0: AMD_NAVI10: bus reset disabled? yes
    [  +0.000004] vfio-pci 0000:04:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
    [  +0.000004] vfio-pci 0000:04:00.0: AMD_NAVI10: performing post-reset
    [  +0.019470] vfio-pci 0000:04:00.0: AMD_NAVI10: reset result = 0
    [  +0.000099] vfio-pci 0000:04:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
    [  +0.000007] vfio-pci 0000:04:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
    [  +0.000003] vfio-pci 0000:04:00.0: vfio_ecap_init: hiding ecap 0x25@0x400
    [  +0.000001] vfio-pci 0000:04:00.0: vfio_ecap_init: hiding ecap 0x26@0x410
    [  +0.000002] vfio-pci 0000:04:00.0: vfio_ecap_init: hiding ecap 0x27@0x440
    [  +1.044135] vfio-pci 0000:08:00.0: vfio_ecap_init: hiding ecap 0x19@0x200
    [  +1.171862] vfio-pci 0000:04:00.0: AMD_NAVI10: version 1.1
    [  +0.000002] vfio-pci 0000:04:00.0: AMD_NAVI10: performing pre-reset
    [  +0.000062] vfio-pci 0000:04:00.0: AMD_NAVI10: performing reset
    [  +0.102421] ATOM BIOS: 115-D199PI0-101
    [  +0.000003] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
    [  +0.163421] vfio-pci 0000:04:00.0: AMD_NAVI10: bus reset disabled? yes
    [  +0.000019] vfio-pci 0000:04:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
    [  +0.000004] vfio-pci 0000:04:00.0: AMD_NAVI10: performing post-reset
    [  +0.017989] vfio-pci 0000:04:00.0: AMD_NAVI10: reset result = 0

ajmadsen, 2021-01-09:
Interesting. It's worth noting that in my case, I couldn't _rebind_ to `snd_hda_intel` after unbinding from vfio, I had to specifically remove the device and rescan the bus. On reinitialization the audio device seemed to recover, and on a subsequent boot of the VM it was available.

ajmadsen, 2021-01-12:
For those of you having the audio reset issue, there's a new issue tracking that bug (as I believe it extends beyond just Navi) at #16. Closing this issue as the OP has stated they had success after using DDU and doing a clean reinstall of AMD drivers.

For those of you with a 5700 XT that is still not working after trying those steps, please open a new issue. I would rather track card quirks in separate issues, and I will close/redirect duplicate issues as I see them.

closed 2021-01-12 by ajmadsen