gnif/vendor-reset #5: Something off with VEGA20 reset error handling

nallar, 2020-11-20:
```
[  183.658047] vfio-pci 0000:43:00.0: AMD_VEGA20: version 1.0
[  183.658052] vfio-pci 0000:43:00.0: AMD_VEGA20: performing pre-reset
[  183.658123] vfio-pci 0000:43:00.0: AMD_VEGA20: performing reset
[  184.390719] vfio-pci 0000:43:00.0: AMD_VEGA20: psp mode1 reset failed
[  184.390723] vfio-pci 0000:43:00.0: AMD_VEGA20: falling back to BACO reset
[  184.603368] vfio-pci 0000:43:00.0: SMU error 0x0
[  184.603371] vfio-pci 0000:43:00.0: Failed to send message 0x25: return 0x0
[  184.603374] vfio-pci 0000:43:00.0: AMD_VEGA20: BACO reset successful
[  184.603376] vfio-pci 0000:43:00.0: AMD_VEGA20: performing post-reset
[  184.622683] vfio-pci 0000:43:00.0: Refused to change power state, currently in D0
[  184.622687] vfio-pci 0000:43:00.0: AMD_VEGA20: reset result = 0
```

Reaches the error here but ends up continuing on to report "BACO reset successful".
https://github.com/gnif/vendor-reset/blob/d88aecfc00fb3b2dccf28a0098de4fc50aae975f/src/amd/common.c#L162

https://github.com/gnif/vendor-reset/blob/2f588b887075860a3a6955fbef09b54e05d4380f/src/amd/vega20.c#L74

Never logs an error here:
https://github.com/gnif/vendor-reset/blob/2f588b887075860a3a6955fbef09b54e05d4380f/src/amd/vega20.c#L172

gnif, 2020-11-21:
Thanks for this, Vega20 is still a WIP and does not correctly reset more than once even on my system here. I am working to resolve this and hope to have it fixed soon.

sirspudd, 2021-01-20:
Top of the morning; I am also attempting to use vendor-reset against a Radeon VII, and am seeing:

```[  119.610696] vfio-pci 0000:0d:00.0: enabling device (0000 -> 0003)
[  119.610782] vfio-pci 0000:0d:00.0: AMD_VEGA20: version 1.0
[  119.610784] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing pre-reset
[  119.610840] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing reset
[  119.826592] vfio-pci 0000:0d:00.0: AMD_VEGA20: no SOL, not attempting BACO reset
[  119.826594] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing post-reset
[  119.837715] vfio-pci 0000:0d:00.0: AMD_VEGA20: reset result = 0
[  119.837814] vfio-pci 0000:0d:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
[  119.837824] vfio-pci 0000:0d:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
[  120.004879] vfio-pci 0000:42:00.0: vfio_ecap_init: hiding ecap 0x19@0x168
[  120.004884] vfio-pci 0000:42:00.0: vfio_ecap_init: hiding ecap 0x1e@0x190
[  120.136914] vfio-pci 0000:0d:00.0: AMD_VEGA20: version 1.0
[  120.136918] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing pre-reset
[  120.136984] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing reset
[  120.352172] vfio-pci 0000:0d:00.0: AMD_VEGA20: no SOL, not attempting BACO reset
[  120.352174] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing post-reset
[  120.363722] vfio-pci 0000:0d:00.0: AMD_VEGA20: reset result = 0
[  123.864692] vfio-pci 0000:0a:00.0: No more image in the PCI ROM
[  123.864774] vfio-pci 0000:0a:00.0: No more image in the PCI ROM
[  131.944035] vfio-pci 0000:0d:00.0: AMD_VEGA20: version 1.0
[  131.944038] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing pre-reset
[  131.944101] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing reset
[  132.160978] vfio-pci 0000:0d:00.0: AMD_VEGA20: no SOL, not attempting BACO reset
[  132.160979] vfio-pci 0000:0d:00.0: AMD_VEGA20: performing post-reset
[  132.173095] vfio-pci 0000:0d:00.0: AMD_VEGA20: reset result = 0```

My VM never actually successfully powers on with vendor-reset in play; dmesg seems to indicate that the module is getting loaded in my initramfs before anything else, so I believe I am not holding it wrong :D I am using this in conjunction with original vfio-pci arguments which should avoid Radeon VII pci initialization; is that a issue?

gnif, 2021-01-22:
I am sorry I have not yet had time to do more work on VEGA20, when things settle a bit I will be back to this.

gnif, 2021-08-04:
Unfortunately we have had zero success in a reset sequence for GPUs based on the VEGA20, as such closing this issue as there is nothing more we can do sorry.

closed 2021-08-04 by gnif