gnif/vendor-reset #8: Vega 8 ?

stuaxo, 2020-12-01:
Any idea how I can find the information needed to add Vega 8 ?

gnif, 2020-12-01:
Vega 8 is `CHIP_RAVEN`, while we could add support for this GPU we can not test it nor do I believe it is useful anyway as iirc you can't use one for VFIO passthrough anyway.

stuaxo, 2020-12-01:
Sure, in this case I'm only interested in being able to reset it, if I do things that stress the GPU then it can crash (things still running but GPU running)... e.g. running a Windows 10 in gnome boxes, at the same time as Qemu with Mac OS 8.

gnif, 2020-12-24:
Closing as out of scope, information on how to reset CHIP_RAVEN can be found by reading the `amdgpu` sources.

closed 2020-12-24 by gnif

stuaxo, 2020-12-24:
Thanks for the info on where to find this.

MatthiasLohr, 2022-01-19:
> Vega 8 is `CHIP_RAVEN`, while we could add support for this GPU we can not test it nor do I believe it is useful anyway as iirc you can't use one for VFIO passthrough anyway.

Why we can't test it and why do you think it is not useful anyway? I'm trying to use if for passthrough, and (as far as I can see for now) resetting is the last problem which stops me from doing that (context: Proxmox VM).

stuaxo, 2022-01-25:
Get enough `VEGA 8` enthusiasts together and maybe we can send a box for testing?

Laptops are about £250 and down second hand at the moment.

cppmonkey, 2022-01-26:
I have an ASROCK X570U4D with a Ryzen 7 5700G.
This includes the AS2500 BMC with VGA output, allowing ProxMox an output for the console.

However due to the IOMMU grouping, the board crashes when the Vega GPU is passed through to a VM.

This hasn't been a problem with the two dedicated Radeon RX460. And there hasn't been any vendor reset issues either.

Does any of your systems have suitable IOMMU grouping to allow for headless use?  If that is even possible

MatthiasLohr, 2022-01-27:
I have a Beelink GTR 5 here. According to the following table (output by Proxmox), theoretically it should be possible:

[image]

What I see is that when trying to passthrough the GPU to a VM, also the SATA "vanishes" from the host. Easily solved by just using this to have an exclusive SSD only for the VM, connected to the then also passed through SATA interfaces.