gnif/vendor-reset #9: Compile error: implicit declaration of function ‘copy_from_user’

kawogi, 2020-12-14:
I made sure the kernel options mentioned in the readme are set, kernel headers are installed as well as `build-essential`.

`sudo dkms install .` results in

```
$ cat /var/lib/dkms/vendor-reset/0.0.18/build/make.log
DKMS make.log for vendor-reset-0.0.18 for kernel 4.19.0-13-amd64 (x86_64)
Mon 14 Dec 2020 06:34:05 AM CET
make -C /lib/modules/4.19.0-13-amd64/build M=/var/lib/dkms/vendor-reset/0.0.18/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-13-amd64'
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/module.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/vendor-reset-dev.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/ioctl.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/ftrace.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/hook.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/common.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/compat.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/firmware.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/navi10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/polaris10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/vega10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/vega20.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/amdgpu_device.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/amdgpu_discovery.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/amdgpu_bios.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/amdgpu_atomfirmware.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/atom.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/common_baco.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/navi10_reg_init.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/navi12_reg_init.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/navi14_reg_init.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/polaris_baco.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/smu7_baco.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/vega10_reg_init.o
  CC [M]  /var/lib/dkms/vendor-reset/0.0.18/build/src/amd/amdgpu/vega20_reg_init.o
/var/lib/dkms/vendor-reset/0.0.18/build/src/ioctl.c: In function ‘vendor_reset_ioctl_reset’:
/var/lib/dkms/vendor-reset/0.0.18/build/src/ioctl.c:34:7: error: implicit declaration of function ‘copy_from_user’; did you mean ‘sg_copy_from_buffer’? [-Werror=implicit-function-declaration]
   if (copy_from_user(&iodev, (void __user *)arg, sizeof(iodev)))
       ^~~~~~~~~~~~~~
       sg_copy_from_buffer
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.19.0-13-common/scripts/Makefile.build:308: /var/lib/dkms/vendor-reset/0.0.18/build/src/ioctl.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-4.19.0-13-common/Makefile:1565: _module_/var/lib/dkms/vendor-reset/0.0.18/build] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-13-amd64'
make: *** [Makefile:8: build] Error 2
```

I also tried `make` which brings up the same error.

I retried `sudo dkms install .` and now I receive:
```
$ sudo dkms install .
Error! DKMS tree already contains: vendor-reset-0.0.18
You cannot add the same module/version combo more than once.
```

So either something worked or there's now a dirty state I'll have to reset somehow.

Are there any prerequisites missing to build the module? Might be due to I'm using LMDE rather than the more common Linux Mint.

Some context:
```
$ uname -a
Linux workstation 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	LMDE 4 (debbie)
Release:	4
Codename:	debbie
```

iggy, 2020-12-14:
My guess is some missing include in that old of a kernel. Maybe try including <linux/uaccess.h> directly in ioctl.c?

kawogi, 2020-12-14:
Thanks, I tried that. `dkms` still looks like it's being blocked by some intermediate state.
```
$ sudo dkms install .
Error! DKMS tree already contains: vendor-reset-0.0.18
You cannot add the same module/version combo more than once.
```

`make` however seems to be sucessful:
```
$ make clean
make -C /lib/modules/4.19.0-13-amd64/build M=/home/kai/vendor-reset clean
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-13-amd64'
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-13-amd64'

$ make
make -C /lib/modules/4.19.0-13-amd64/build M=/home/kai/vendor-reset modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-13-amd64'
  CC [M]  /home/kai/vendor-reset/src/module.o
  CC [M]  /home/kai/vendor-reset/src/vendor-reset-dev.o
  CC [M]  /home/kai/vendor-reset/src/ioctl.o
  CC [M]  /home/kai/vendor-reset/src/ftrace.o
  CC [M]  /home/kai/vendor-reset/src/hook.o
  CC [M]  /home/kai/vendor-reset/src/amd/common.o
  CC [M]  /home/kai/vendor-reset/src/amd/compat.o
  CC [M]  /home/kai/vendor-reset/src/amd/firmware.o
  CC [M]  /home/kai/vendor-reset/src/amd/navi10.o
  CC [M]  /home/kai/vendor-reset/src/amd/polaris10.o
  CC [M]  /home/kai/vendor-reset/src/amd/vega10.o
  CC [M]  /home/kai/vendor-reset/src/amd/vega20.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/amdgpu_device.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/amdgpu_discovery.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/amdgpu_bios.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/amdgpu_atomfirmware.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/atom.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/common_baco.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/navi10_reg_init.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/navi12_reg_init.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/navi14_reg_init.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/polaris_baco.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/smu7_baco.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/vega10_reg_init.o
  CC [M]  /home/kai/vendor-reset/src/amd/amdgpu/vega20_reg_init.o
  LD [M]  /home/kai/vendor-reset/vendor-reset.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/kai/vendor-reset/vendor-reset.mod.o
  LD [M]  /home/kai/vendor-reset/vendor-reset.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-13-amd64'
```
`make install` however fails differently:
```
$ sudo make install
make -C /lib/modules/4.19.0-13-amd64/build M= INSTALL_MOD_PATH= modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-13-amd64'
cp: cannot stat './modules.order': No such file or directory
make[3]: *** [/usr/src/linux-headers-4.19.0-13-common/Makefile:1304: _modinst_] Error 1
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-13-amd64'
make: *** [Makefile:11: install] Error 2
```

The mentioned file doesn't exist from my perspective as well.

This is all new to me so maybe I'm making some stupid mistake.

Would a kernel upgrade help? Any minimum version recommended?

kawogi, 2020-12-14:
Ok I upgraded to 5.9 and I'm no longer able to boot the machine. Looks like nVidia kernel modules messed things up :-1: 
Old kernel doesn't boot any more as well. Also some nVidia error messages. Might take some time to figure that out.

Ansa89, 2020-12-14:
To compile `vendor-reset` on old kernels, you might need this (ugly) patch:
```
--- a/src/ioctl.c
+++ b/src/ioctl.c
@@ -21,6 +21,7 @@ Place, Suite 330, Boston, MA 02111-1307 USA
 #include "vendor-reset-ioctl.h"
 
 #include <linux/miscdevice.h>
+#include <linux/uaccess.h>
 
 #define VENDOR_RESET_IOCTL_DEVNAME "vendor_reset"
 
```

kawogi, 2020-12-14:
That's what @iggy suggested. I had a lot of follow up problems, so I'm currently rebuilding the machine using the latest Manjaro (minimal). Will report back.

kawogi, 2020-12-16:
Cannot reproduce the problem with a more recent kernel and build environment.

closed 2020-12-16 by kawogi