gnif/vendor-reset #11: Consider upstreaming these quirks

bjorn-helgaas, 2020-12-23:
The readme says "while it would be great to have these in the kernel as PCI quirks, some of the reset procedures are very complex and would never be accepted as a quirk."  But I don't remember seeing a discussion about adding these to the kernel.

The existing user experience (seeing reset failures, VFIO passthrough issues, etc, then stumbling over this out-of-tree workaround) seems pretty poor.  Maybe we should have a conversation about how to improve that?  linux-pci@vger.kernel.org would be a good place to start.

gnif, 2020-12-23:
I am sorry but somehow I think it's completely pointless to do so. Not only are these routines still incomplete and do not always work, but they also consist of thousands of lines of code pulled from `amdgpu` and modified to perform the required setup, such as ATOM bios parsing and register offset discovery. The bloat and code duplication it would add to the kernel would be thrown out in an instant.

bjorn-helgaas, 2020-12-23:
Well, it certainly won't happen if you don't try.  Bloat and code duplication sound like things that could possibly be solved via some refactoring.

My main concern is the user experience.  If this problem doesn't affect very many people, maybe it's not worth worrying about.  I don't know how to guess, but I found this via https://www.reddit.com/r/VFIO/comments/ki4kyu/does_the_rx_580_still_have_the_reset_bug/, where a dozen people commented in a single day.

gnif, 2020-12-24:
You would have to provide some means for the pci quick to be able to call into `amdgpu` code directly without having the module loaded and ignoring all the state & setup information the driver stores and retains. To make this possible AMD would need to massively refactor their own code and be willing to implement and maintain these exports for quirk usage.

These GPUs never advertised FLR nor were they intended to be used for VFIO passthrough. We are attempting to use these GPUs in a manner in which they were not designed to be used. As such, these issues are NOT bugs, and a quirk doesn't even make sense. For people using these GPUs as intended, the `amdgpu` module already performs everything needed to correctly use these GPUs.

I have spent three years now, countless hours, pleading with AMD and negotiating with them on how to best address these issues and learning the amdgpu code base. I have been told directly that there is no incentive to fix this in older generations (pre BigNavi) and efforts have been made and will continue to be made to ensure that hardware reset support will function in future generations, however, it is still not an officially supported feature of the GPU.

closed 2020-12-24 by gnif

bobafetthotmail, 2020-12-25:
> 
> My main concern is the user experience. 

Installing this module works similarly to installing NVIDIA proprietary driver, or any non-upstream wireless dongle drivers, or ZFS filesystem. It's not particularly difficult, you just need to install the basic development packages for your distro and run the command in the console to have all the work done automatically.

bjorn-helgaas, 2020-12-29:
Ideally users wouldn't have to install the module at all.  But that's not what I'm talking about here.  I'm talking about how users get from "something is broken" to "the workaround is to install module X."  Is that a trivial step?

bobafetthotmail, 2020-12-29:
> Ideally users wouldn't have to install the module at all.

pfft, Torvalds will never accept a large dirty hack like this, and as Gnif also said the only way to make this actually work without a full team of paid developers with access to AMD private information is doing dirty hacks.
There are plenty of cases where people install external modules like this for edge cases.

>But that's not what I'm talking about here. I'm talking about how users get from "something is broken" to "the workaround is to install module X." Is that a trivial step?

I'd say it is good enough for what this is. We are talking about VFIO-based GPU passthrough, which is far from trivial even without this issue and requires a lengthy process of manual editing of system config files to even get to the point where you need this module.

Anybody that gets to the point of experiencing this issue at all will be following a tutorial from a KVM enthusiast community where this module is known, it started years ago as a kernel patch and it got some recognition but as the hacks grew it became apparent that upstreaming it would not work.

Both Gnif and this module are well-known on the Reddit section about gpu passthrough, https://www.reddit.com/r/VFIO/comments/kj0hoh/how_to_install_kernel_options_of_gnifs_vendorreset/
on Level1Tech forums (another major gpu passthrough community with tutorials and whatnot) he is a rock star as that's where this all started, and he posted a thread about this module as well https://forum.level1techs.com/t/amd-polaris-vega-navi-reset-project-vendor-reset/163801
on both these community anybody searching or asking questions about the issues will get answers to come here.

Someone posted blog how to use this with Proxmox virtualization distro https://www.nicksherlock.com/2020/11/working-around-the-amd-gpu-reset-bug-on-proxmox/

Arch Linux AUR (community package repository) has a few packages at various versions for this module https://aur.archlinux.org/packages/?O=0&SeB=nd&K=vendor-reset&outdated=&SB=n&SO=a&PP=50&do_Search=Go

And all this happened within a couple months this project started (first commit 31 october).