gnif/vendor-reset #12: xfx pro thicc 3 5600 xt navi and audio

Pale091, 2020-12-27:
Just discovered this vendor-reset module today. Tried your reset kernel patches in the past with never any luck. Only problem I have with this vendor-reset is that the hdmi audio for the card goes away or doesn't reset properly after one restart, shutdown. I have to restart the host to get it back.

Gentoo. 1900x threadripper, xfx pro thicc 3 5600 xt navi, asrock-x399-phantom-gaming-6.

I'm gonna try disabling acs in the uefi bios and see if that helps.
Appreciate all your hard work.

```

IOMMU Group 32:
	44:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] [1002:731f] (rev ca)
IOMMU Group 33:
	44:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 HDMI Audio [1002:ab38]
```

Maybe cause my audio and vga are in different groups? It works perfect the first time.

```
[ 4088.164275] vfio-pci 0000:44:00.1: disabling bus mastering
[ 4088.164355] vfio-pci 0000:44:00.1: restoring config space at offset 0x4 (was 0x100400, writing 0x100002)
[ 4088.175538] vfio-pci 0000:44:00.0: disabling bus mastering
[ 4088.175656] vfio-pci 0000:44:00.0: AMD_NAVI10: version 1.1
[ 4088.175657] vfio-pci 0000:44:00.0: AMD_NAVI10: performing pre-reset
[ 4088.175677] vfio-pci 0000:44:00.0: saving config space at offset 0x0 (reading 0x731f1002)
[ 4088.175679] vfio-pci 0000:44:00.0: saving config space at offset 0x4 (reading 0x100400)
[ 4088.175681] vfio-pci 0000:44:00.0: saving config space at offset 0x8 (reading 0x30000ca)
[ 4088.175683] vfio-pci 0000:44:00.0: saving config space at offset 0xc (reading 0x800010)
[ 4088.175685] vfio-pci 0000:44:00.0: saving config space at offset 0x10 (reading 0xc000000c)
[ 4088.175687] vfio-pci 0000:44:00.0: saving config space at offset 0x14 (reading 0x7fc)
[ 4088.175688] vfio-pci 0000:44:00.0: saving config space at offset 0x18 (reading 0xd000000c)
[ 4088.175690] vfio-pci 0000:44:00.0: saving config space at offset 0x1c (reading 0x7fc)
[ 4088.175692] vfio-pci 0000:44:00.0: saving config space at offset 0x20 (reading 0x3001)
[ 4088.175694] vfio-pci 0000:44:00.0: saving config space at offset 0x24 (reading 0x82300000)
[ 4088.175696] vfio-pci 0000:44:00.0: saving config space at offset 0x28 (reading 0x0)
[ 4088.175697] vfio-pci 0000:44:00.0: saving config space at offset 0x2c (reading 0x57101682)
[ 4088.175698] vfio-pci 0000:44:00.0: saving config space at offset 0x30 (reading 0x82380000)
[ 4088.175700] vfio-pci 0000:44:00.0: saving config space at offset 0x34 (reading 0x48)
[ 4088.175701] vfio-pci 0000:44:00.0: saving config space at offset 0x38 (reading 0x0)
[ 4088.175702] vfio-pci 0000:44:00.0: saving config space at offset 0x3c (reading 0x1ff)
[ 4088.175750] vfio-pci 0000:44:00.0: AMD_NAVI10: performing reset
[ 4088.275033] ATOM BIOS: 113-170WBNAVIXLE6GB_MIC_191122_W8
[ 4088.275036] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[ 4088.275039] vfio-pci 0000:44:00.0: AMD_NAVI10: bus reset disabled? yes
[ 4088.275043] vfio-pci 0000:44:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: fda1001, mp1 intr enabled? yes, bl ready? yes
[ 4088.275044] vfio-pci 0000:44:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[ 4088.275086] vfio-pci 0000:44:00.0: AMD_NAVI10: begin psp mode 1 reset
[ 4088.275089] vfio-pci 0000:44:00.0: saving config space at offset 0x0 (reading 0x731f1002)
[ 4088.275090] vfio-pci 0000:44:00.0: saving config space at offset 0x4 (reading 0x100402)
[ 4088.275091] vfio-pci 0000:44:00.0: saving config space at offset 0x8 (reading 0x30000ca)
[ 4088.275093] vfio-pci 0000:44:00.0: saving config space at offset 0xc (reading 0x800010)
[ 4088.275094] vfio-pci 0000:44:00.0: saving config space at offset 0x10 (reading 0xc000000c)
[ 4088.275095] vfio-pci 0000:44:00.0: saving config space at offset 0x14 (reading 0x7fc)
[ 4088.275097] vfio-pci 0000:44:00.0: saving config space at offset 0x18 (reading 0xd000000c)
[ 4088.275098] vfio-pci 0000:44:00.0: saving config space at offset 0x1c (reading 0x7fc)
[ 4088.275099] vfio-pci 0000:44:00.0: saving config space at offset 0x20 (reading 0x3001)
[ 4088.275100] vfio-pci 0000:44:00.0: saving config space at offset 0x24 (reading 0x82300000)
[ 4088.275102] vfio-pci 0000:44:00.0: saving config space at offset 0x28 (reading 0x0)
[ 4088.275103] vfio-pci 0000:44:00.0: saving config space at offset 0x2c (reading 0x57101682)
[ 4088.275104] vfio-pci 0000:44:00.0: saving config space at offset 0x30 (reading 0x82380000)
[ 4088.275106] vfio-pci 0000:44:00.0: saving config space at offset 0x34 (reading 0x48)
[ 4088.275107] vfio-pci 0000:44:00.0: saving config space at offset 0x38 (reading 0x0)
[ 4088.275108] vfio-pci 0000:44:00.0: saving config space at offset 0x3c (reading 0x1ff)
[ 4088.779533] vfio-pci 0000:44:00.0: AMD_NAVI10: mode1 reset succeeded
[ 4088.779585] vfio-pci 0000:44:00.0: restoring config space at offset 0x30 (was 0x0, writing 0x82380000)
[ 4088.779589] vfio-pci 0000:44:00.0: restoring config space at offset 0x24 (was 0x0, writing 0x82300000)
[ 4088.779592] vfio-pci 0000:44:00.0: restoring config space at offset 0x20 (was 0x1, writing 0x3001)
[ 4088.779595] vfio-pci 0000:44:00.0: restoring config space at offset 0x1c (was 0x0, writing 0x7fc)
[ 4088.779598] vfio-pci 0000:44:00.0: restoring config space at offset 0x18 (was 0xc, writing 0xd000000c)
[ 4088.779601] vfio-pci 0000:44:00.0: restoring config space at offset 0x14 (was 0x0, writing 0x7fc)
[ 4088.779604] vfio-pci 0000:44:00.0: restoring config space at offset 0x10 (was 0xc, writing 0xc000000c)
[... 145 lines cut ...]
[ 4204.971643] vfio-pci 0000:44:00.0: saving config space at offset 0x10 (reading 0xc000000c)
[ 4204.971644] vfio-pci 0000:44:00.0: saving config space at offset 0x14 (reading 0x7fc)
[ 4204.971646] vfio-pci 0000:44:00.0: saving config space at offset 0x18 (reading 0xd000000c)
[ 4204.971647] vfio-pci 0000:44:00.0: saving config space at offset 0x1c (reading 0x7fc)
[ 4204.971648] vfio-pci 0000:44:00.0: saving config space at offset 0x20 (reading 0x3001)
[ 4204.971649] vfio-pci 0000:44:00.0: saving config space at offset 0x24 (reading 0x82300000)
[ 4204.971651] vfio-pci 0000:44:00.0: saving config space at offset 0x28 (reading 0x0)
[ 4204.971652] vfio-pci 0000:44:00.0: saving config space at offset 0x2c (reading 0x57101682)
[ 4204.971653] vfio-pci 0000:44:00.0: saving config space at offset 0x30 (reading 0x82380000)
[ 4204.971655] vfio-pci 0000:44:00.0: saving config space at offset 0x34 (reading 0x48)
[ 4204.971656] vfio-pci 0000:44:00.0: saving config space at offset 0x38 (reading 0x0)
[ 4204.971657] vfio-pci 0000:44:00.0: saving config space at offset 0x3c (reading 0x1ff)
[ 4204.971702] vfio-pci 0000:44:00.0: AMD_NAVI10: performing reset
[ 4205.070265] ATOM BIOS: 113-170WBNAVIXLE6GB_MIC_191122_W8
[ 4205.070266] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[ 4205.285843] vfio-pci 0000:44:00.0: AMD_NAVI10: bus reset disabled? yes
[ 4205.285848] vfio-pci 0000:44:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[ 4205.285851] vfio-pci 0000:44:00.0: AMD_NAVI10: performing post-reset
[ 4205.285912] vfio-pci 0000:44:00.0: restoring config space at offset 0x1c (was 0x0, writing 0x7fc)
[ 4205.285915] vfio-pci 0000:44:00.0: restoring config space at offset 0x18 (was 0xc, writing 0xd000000c)
[ 4205.285918] vfio-pci 0000:44:00.0: restoring config space at offset 0x14 (was 0x0, writing 0x7fc)
[ 4205.285921] vfio-pci 0000:44:00.0: restoring config space at offset 0x10 (was 0xc, writing 0xc000000c)
[ 4205.285925] vfio-pci 0000:44:00.0: restoring config space at offset 0x4 (was 0x100402, writing 0x100400)
[ 4205.297527] vfio-pci 0000:44:00.0: AMD_NAVI10: reset result = 0
```

Pale091, 2020-12-27:
I dropped softdep snd_hda_intel pre: vfio-pci from /etc/modprobe.d to see if it made a difference. I can still restart forever without issue, but if I shutdown I lose my hdmi audio until I shutdown the the host.
https://pastebin.com/naPWK67H  a proper dmesg.

Pale091, 2020-12-27:
Well this is pretty awesome. Looks like all I have to do when the audio goes down is rescan the bus as described here: https://old.reddit.com/r/VFIO/comments/jjy74y/navi_reset_bug_kernel_patch_v2/gahsyeb/

Thank you so much for you work on this!!!

gnif, 2020-12-29:
no worries, but for future reference (and others) this module does not attempt to reset the audio device, usually it's not needed, nor do we have any information on how to reset it. As such it's out of scope.

closed 2020-12-29 by gnif