gnif/vendor-reset #13: Initializing fans

9ary, 2020-12-28:
The card I pass through (RX550 Pulse) is rather noisy out of reset. The card is left in that state after every host reboot and resuming from S3 (the GOP/VBIOS does turn it down significantly, but not to the point of being silent, and it doesn't run during resume).
Letting amdgpu bind to the card does allow the fan to quiet down, but it complains very loud (in dmesg) when detaching and booting the VM, and probably causes more harm that isn't detectable at first glance, so I'd rather not do that. This also leaves the card in a non-functional state for Windows when not using vendor-reset.
Booting the VM and then shutting it down leaves the fan in the quiet state, as long as vendor-reset is not loaded. Otherwise the card resets immediately after shutdown, and the fan gets loud again.

Is it possible for vendor-reset to initialize the fan controller? Possibly even enough to get zero-RPM mode working?

gnif, 2020-12-29:
No, it's a `reset`, it's supposed to put the GPU into the pre-boot uninitialized state, that's is the entire purpose of this module.

closed 2020-12-29 by gnif

9ary, 2020-12-29:
How else would you suggest handling this then? You said it yourself, this module contains a large amount of code copy-pasted from amdgpu, and I don't think creating a third codebase that duplicates the same code for a minor QoL improvement would be very productive. I just want to turn the fan off, and I don't want to have a VM constantly running for that.

gnif, 2020-12-29:
Perhaps a userspace tool. You're missing the point of this project though, it's not to put the GPU into a low power state, it's to prepare the GPU for re-use in a VM where the VM's driver will handle the specifics. If we init anything, when the VM tries to use it, it will not work. There is no way to detect VM shutdown/start vs VM reset.

9ary, 2020-12-29:
I understand the purpose of this module, there's no need to be so aggressive.

I don't really know what it takes to initialize the fans, but I do know that the VBIOS does it with no impact on the VM's ability to initialize the GPU. In fact, I don't have the "reset bug", my VM can boot and reboot as many times as I want without rebooting the host, as long as I don't let amdgpu touch the passthrough card (whether it's the host or a Linux VM). When shutting down, Windows leaves the fan off, and then I can start the VM again just fine.

Considering this, I thought it wouldn't be a problem to change the fan speed during reset, since the VBIOS obviously does it without loading firmware or initializing the GPU, it only does enough to get VGA emulation or UEFI GOP working during boot. I don't really want to put the GPU in a low power state, proper power management is much more complex and clearly out of scope. But doing something the VBIOS does didn't seem unreasonable to me. It can even be implemented behind a module parameter, which means it wouldn't break anything for anyone who doesn't touch it.

The only thing vendor-reset seems to add for me is the ability to boot a Windows VM after amdgpu has touched the card which I do not need. Unless of course I let amdgpu bind to the card to turn off the fan, but like I said, this doesn't seem like a good idea.

Anyway, thanks for answering my question, and sorry for wasting your time.

gnif, 2020-12-29:
I am sorry I did not intend to come across as aggressive. We do not re-post the VBIOS, we are returning the GPU to a state pre-system boot so the VM's BIOS can post the GPU's BIOS.