gnif/vendor-reset #16: Audio device fails after first reset

ajmadsen, 2021-01-12:
After the first reset, HDMI audio devices usually crash and show up as unrecoverable in the Windows device manager. This has been reported with other OSes as well.

netinetwalker, 2021-01-13:
I have the branch with proxmox and a 5700xt.
I must remove the device in windows manuell to get it autodetected after the next boot.
After that no problems with the audiodevice anymore. 

Thanks for the fix. 
Five restarts without any issues

giganode, 2021-01-13:
Still having the broken audio. Works as before for me. Restarts are fine, but once I shut the vm down audio never comes back.
Did 2 soft restarts before going further. Here is my output:

[  969.437427] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[  969.437428] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[  969.437517] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[  969.792680] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[  969.792686] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  969.792691] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[  969.804298] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[  970.158404] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[  970.158407] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[  970.158502] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[  970.490677] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[  970.490682] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  970.490684] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[  970.502294] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[ 1326.236247] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 1326.236249] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[ 1326.236330] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[ 1326.341045] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1326.341049] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 38901c42, mp1 intr enabled? yes, bl ready? yes
[ 1326.341050] vfio-pci 0000:2f:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[ 1326.341161] vfio-pci 0000:2f:00.0: AMD_NAVI10: begin psp mode 1 reset
[ 1326.847460] vfio-pci 0000:2f:00.0: AMD_NAVI10: mode1 reset succeeded
[ 1328.614444] vfio-pci 0000:2f:00.0: AMD_NAVI10: PSP mode1 reset successful
[ 1328.614452] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[ 1328.626450] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[ 1422.491192] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 1422.491194] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[ 1422.491275] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[ 1422.596130] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1422.596135] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 4c98cc9a, mp1 intr enabled? yes, bl ready? yes
[ 1422.596135] vfio-pci 0000:2f:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[ 1422.596273] vfio-pci 0000:2f:00.0: AMD_NAVI10: begin psp mode 1 reset
[ 1423.101394] vfio-pci 0000:2f:00.0: AMD_NAVI10: mode1 reset succeeded
[ 1424.869496] vfio-pci 0000:2f:00.0: AMD_NAVI10: PSP mode1 reset successful
[ 1424.869507] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[ 1424.881494] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[ 1517.316999] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 1517.317002] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[ 1517.317089] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[ 1517.422002] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1517.422007] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 51eed70d, mp1 intr enabled? yes, bl ready? yes
[ 1517.422009] vfio-pci 0000:2f:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[ 1517.422198] vfio-pci 0000:2f:00.0: AMD_NAVI10: begin psp mode 1 reset
[ 1517.924360] vfio-pci 0000:2f:00.0: AMD_NAVI10: mode1 reset succeeded
[ 1519.693335] vfio-pci 0000:2f:00.0: AMD_NAVI10: PSP mode1 reset successful
[ 1519.693341] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[ 1519.705343] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[ 1677.947747] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 1677.947748] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[ 1677.947834] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[ 1678.310348] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1678.310356] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[ 1678.310361] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[ 1678.322631] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[ 1678.670721] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 1678.670724] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[ 1678.670820] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[ 1679.019022] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1679.019028] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[ 1679.019031] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[ 1679.030619] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0
[ 1874.089235] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 1874.089238] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[ 1874.089325] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing reset
[ 1874.194283] vfio-pci 0000:2f:00.0: AMD_NAVI10: bus reset disabled? yes
[ 1874.194288] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 5f9a82fb, mp1 intr enabled? yes, bl ready? yes
[ 1874.194289] vfio-pci 0000:2f:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[... 87 lines cut ...]
[ 2684.629245] vfio-pci 0000:2f:00.0: AMD_NAVI10: version 1.1
[ 2684.629246] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing pre-reset
[... cut ...]
[ 2842.061012] vfio-pci 0000:2f:00.0: AMD_NAVI10: SMU response reg: 1, sol reg: 97fd6716, mp1 intr enabled? yes, bl ready? yes
[ 2842.061014] vfio-pci 0000:2f:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[ 2842.061120] vfio-pci 0000:2f:00.0: AMD_NAVI10: begin psp mode 1 reset
[ 2842.564047] vfio-pci 0000:2f:00.0: AMD_NAVI10: mode1 reset succeeded
[ 2844.436019] vfio-pci 0000:2f:00.0: AMD_NAVI10: PSP mode1 reset successful
[ 2844.436026] vfio-pci 0000:2f:00.0: AMD_NAVI10: performing post-reset
[ 2844.448020] vfio-pci 0000:2f:00.0: AMD_NAVI10: reset result = 0

ajmadsen, 2021-01-13:
@netinetwalker 
> I must remove the device in windows manuell to get it autodetected after the next boot.
> After that no problems with the audiodevice anymore.

Is this every time you reboot the host, or was this just a one-time thing that happened right after you installed this version?

@giganode
> Restarts are fine, but once I shut the vm down audio never comes back.

This is the issue I was experiencing as well with my hardware, so I would think this would fix it for you too. How are you installing the module? Are you sure the new version of the module got loaded before you tested?

giganode, 2021-01-13:
@ajmadsen just reconized master was reverted..

ajmadsen, 2021-01-13:
No worries, that was just a miscommunication as I didn't consider the PR ready for merge yet. I suggest building from the branch, as I'll do any further dev work there until it gets merged.

giganode, 2021-01-13:
@ajmadsen I built again.

> Cloning into 'vendor-reset'...
> Switched to a new branch 'feature/audio_reset'

Host is running Unraid 6.9.0-rc2 with kernel version 5.10.1.

I can now boot between Windows 10 20H2, macOS Big Sur 11.1 and Ubuntu 20.10! 

Thank you all very much for your work! No broken audio until now! Now I can finally forget the navi patch ^.^

netinetwalker, 2021-01-14:
@ajmadsen after some reboots of the vm the problem is comming back. 
But after a host restart it works for some time. I think the audio device survives a windows reboot inside the vm but not a complete vm stop and start? If windows detects a device problem i must remove it from the device manager and reboot the vm to get it back working.

ajmadsen, 2021-01-14:
@netinetwalker Interesting, I didn't run into that issue. Reboots worked fine even before the fix, but the fix should keep the device from dying on vm stop and start. I didn't test removing the device when it failed and rebooting the vm, so I'm not sure if that behavior also worked prior to the fix or not.

giganode, 2021-01-14:
@netinetwalker What happens if you just do shutdowns? Have you tried different OSes?

derpuma, 2021-01-14:
@ajmadsen @gnif I want to shout out a BIG THANK YOU as finally the latest vendor-reset feature/audio_reset #18 is also working for my Powercolor 5700XT now! The Navi-Patch never did the job for me with this card and vendor-reset patch only worked for my RX 480 GPUs, so I think it must have been something to do with the gpu audio in case of the 5700 XT.
Again, thanks so much!

netinetwalker, 2021-01-15:
@giganode its broken after a shutdown
i think it only survives if i reboot the vm from inside the vm.?

giganode, 2021-01-15:
@netinetwalker I can confirm 3 working navi cards on 3 different unraid systems, now.
Yesterday, @derpuma even tried to force stop a vm and could boot a vm after that.
Has to be something specific with your system....

ajmadsen, 2021-01-15:
@netinetwalker it sounds like you may not be running the audio fix branch then, as that is the same behavior as prior to the new attempted fix.

netinetwalker, 2021-01-16:
ok i think it was the 0.0.18 module ... in modinfo i cannot see any version number.

dkms status “WARNING! Diff between built and installed module”

dkms remove vendor-reset/0.0.18 --all
dkms remove vendor-reset/0.0.19 --all
dkms build vendor-reset/0.0.19
dkms install vendor-reset/0.0.19

reboot

ajmadsen, 2021-01-16:
@netinetwalker after reinstalling did you find success?

netinetwalker, 2021-01-16:
@ajmadsen two reboots outside the vm and it looks good at the moment

ajmadsen, 2021-01-23:
In master as of 24c9fc547366e13a96ad497b038d1c378423a39c

PR #17 (2021-01-13): Fix audio function dying on GPU reset
Fixes #16
closed 2021-01-23 by ajmadsen