gnif/vendor-reset #21: Not working with 5700xt Pulse

lbibass, 2021-01-30:
I'm having an issue with vendor-reset on my current arch machine. 

I'm using kernel 5.10.10-115-tkg-bmq. 

I am able to detach and re-attach my card a few times, but then I start getting an error in dmesg. 
```
[ 1078.072222] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:03.1/0000:07:00.0/0000:08:00.0/0000:09:00.0/vbios_version'

Any ideas on what is going on?

lbibass, 2021-01-30:
Update: if I just detach and re-attach the gpu without doing anything else, other than removing the vfio drivers, I get this error:

```[  271.451374] amdgpu 0000:09:00.0: vgaarb: deactivate vga console
[  271.451444] [drm] initializing kernel modesetting (NAVI10 0x1002:0x731F 0x1DA2:0xE411 0xC1).
[  271.451446] amdgpu 0000:09:00.0: amdgpu: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
[  271.451459] [drm] register mmio base: 0xFCD00000
[  271.451459] [drm] register mmio size: 524288
[  271.453509] [drm] add ip block number 0 <nv_common>
[  271.453510] [drm] add ip block number 1 <gmc_v10_0>
[  271.453510] [drm] add ip block number 2 <navi10_ih>
[  271.453512] [drm] add ip block number 3 <psp>
[  271.453512] [drm] add ip block number 4 <smu>
[  271.453513] [drm] add ip block number 5 <dm>
[  271.453514] [drm] add ip block number 6 <gfx_v10_0>
[  271.453514] [drm] add ip block number 7 <sdma_v5_0>
[  271.453515] [drm] add ip block number 8 <vcn_v2_0>
[  271.453515] [drm] add ip block number 9 <jpeg_v2_0>
[  271.453529] amdgpu 0000:09:00.0: amdgpu: Fetched VBIOS from VFCT
[  271.453530] amdgpu: ATOM BIOS: 113-1E4112U-O45
[  271.453534] [drm] VCN decode is enabled in VM mode
[  271.453535] [drm] VCN encode is enabled in VM mode
[  271.453535] [drm] JPEG decode is enabled in VM mode
[  271.453540] amdgpu 0000:09:00.0: amdgpu: MODE1 reset
[  271.453611] amdgpu 0000:09:00.0: amdgpu: GPU psp mode1 reset
[  271.965668] [drm] psp mode1 reset succeed 
[  271.965781] [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
[  271.965788] amdgpu 0000:09:00.0: amdgpu: VRAM: 8176M 0x0000008000000000 - 0x00000081FEFFFFFF (8176M used)
[  271.965789] amdgpu 0000:09:00.0: amdgpu: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
[  271.965801] [drm] Detected VRAM RAM=8176M, BAR=8192M
[  271.965802] [drm] RAM width 256bits GDDR6
[  271.965894] [TTM] Zone  kernel: Available graphics memory: 16393472 KiB
[  271.965894] [TTM] Zone   dma32: Available graphics memory: 2097152 KiB
[  271.965895] [TTM] Initializing pool allocator
[  271.965898] [TTM] Initializing DMA pool allocator
[  271.965944] [drm] amdgpu: 8176M of VRAM memory ready
[  271.965946] [drm] amdgpu: 8176M of GTT memory ready.
[  271.965947] [drm] GART: num cpu pages 131072, num gpu pages 131072
[  271.966102] [drm] PCIE GART of 512M enabled (table at 0x0000008000000000).
[  271.967371] [drm] Found VCN firmware Version ENC: 1.10 DEC: 5 VEP: 0 Revision: 13
[  271.967373] [drm] PSP loading VCN firmware
[  272.691999] [drm:psp_v11_0_ring_create [amdgpu]] *ERROR* Failed to wait for sOS ready for ring creation
[  272.692042] [drm:psp_hw_start [amdgpu]] *ERROR* PSP create ring failed!
[  272.692082] [drm:psp_hw_init [amdgpu]] *ERROR* PSP firmware loading failed
[  272.692120] [drm:amdgpu_device_fw_loading [amdgpu]] *ERROR* hw_init of IP block <psp> failed -22
[  272.692120] amdgpu 0000:09:00.0: amdgpu: amdgpu_device_ip_init failed
[  272.692124] amdgpu 0000:09:00.0: amdgpu: Fatal error during GPU init
[  272.692143] amdgpu: probe of 0000:09:00.0 failed with error -22```

johncolby, 2021-02-05:
FWIW:
* I have this card and it is working fine with vendor-reset, so your card hardware at least is OK (in general).
* That audio error should be incidental/nonblocking.

When you say detach/reattach, are you talking about GPU reset upon guest restart? Or amdgpu and vfio-pci stub drivers on the host? Just to confirm, you *can* successfully pass the card through to your guest at least once, correct?

A couple random ideas:
* Have you tried with other kernels?
* Have you tried with blacklisting the GPU on host boot?
* Have you tried detaching/reattaching the amdgpu driver in the host, without vendor-reset installed, to confirm that works?
* Have you tried manually providing a GPU vBIOS with libvirt?

kuviman, 2021-03-21:
Having similar issues

@johncolby 

> Have you tried with other kernels?

same thing on lts/zen/default arch kernels

> Have you tried with blacklisting the GPU on host boot?

this is the only thing that got passthrough to work, in combination with `video=efifb:off`, I can restart VM and passthrough again if I never loaded amdgpu. I can load amdgpu once after using VM and use host, but then im having issues if I want to passthrough again.

> Have you tried detaching/reattaching the amdgpu driver in the host, without vendor-reset installed, to confirm that works?

This one is the issue I guess. With or without vendor-reset, when I detach amdgpu and then try to reattach it, I get this error:

```
[drm:psp_v11_0_ring_create [amdgpu]] *ERROR* Failed to wait for sOS ready for ring creation
[drm:psp_hw_start [amdgpu]] *ERROR* PSP create ring failed!
[drm:psp_hw_init [amdgpu]] *ERROR* PSP firmware loading failed
[amdgpu]] *ERROR* hw_init of IP block <psp> failed -22
amdgpu 0000:09:00.0: amdgpu: amdgpu_device_ip_init failed
amdgpu 0000:09:00.0: amdgpu: Fatal error during GPU init
amdgpu: probe of 0000:09:00.0 failed with error -22
```

gnif, 2021-03-21:
https://github.com/gnif/vendor-reset/issues/25#issuecomment-799838629

closed 2021-03-21 by gnif