gnif/vendor-reset #23: Navi14 W5500 support

thielemans, 2021-02-01:
I am trying to have GPU passthrough of a W5500. Passing through works on the first time a Linux guest VM is launched but fails after that VM is shut down an started again. The host is CentOS 8.3 running 4.18.0-240.10.1. I am aware this specific GPU is not in the supported devices list, but what could be done to include it?

First time VM launch
`[  116.892875] vfio-pci 0000:86:00.0: enabling device (0142 -> 0143)
[  116.893018] vfio-pci 0000:86:00.0: AMD_NAVI14: version 1.1
[  116.893019] vfio-pci 0000:86:00.0: AMD_NAVI14: performing pre-reset
[  116.905913] vfio-pci 0000:86:00.0: AMD_NAVI14: performing reset
[  116.942304] ATOM BIOS: 113-D3250100-102
[  116.942306] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  117.183237] vfio-pci 0000:86:00.0: AMD_NAVI14: bus reset disabled? yes
[  117.183243] vfio-pci 0000:86:00.0: AMD_NAVI14: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  117.183247] vfio-pci 0000:86:00.0: AMD_NAVI14: performing post-reset
[  117.206872] vfio-pci 0000:86:00.0: AMD_NAVI14: reset result = 0
[  117.207039] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x19@0x270
[  117.207051] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x1b@0x2d0
[  117.207056] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x25@0x400
[  117.207057] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x26@0x410
[  117.207058] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x27@0x440
[  117.241993] vfio-pci 0000:86:00.0: AMD_NAVI14: version 1.1
[  117.241995] vfio-pci 0000:86:00.0: AMD_NAVI14: performing pre-reset
[  117.253918] vfio-pci 0000:86:00.0: AMD_NAVI14: performing reset
[  117.291160] ATOM BIOS: 113-D3250100-102
[  117.291162] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  117.531555] vfio-pci 0000:86:00.0: AMD_NAVI14: bus reset disabled? yes
[  117.531561] vfio-pci 0000:86:00.0: AMD_NAVI14: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  117.531565] vfio-pci 0000:86:00.0: AMD_NAVI14: performing post-reset
[  117.554883] vfio-pci 0000:86:00.0: AMD_NAVI14: reset result = 0`

First time VM shutdown:
`[  244.026532] vfio-pci 0000:86:00.0: AMD_NAVI14: version 1.1
[  244.026534] vfio-pci 0000:86:00.0: AMD_NAVI14: performing pre-reset
[  244.038373] vfio-pci 0000:86:00.0: AMD_NAVI14: performing reset
[  244.075239] ATOM BIOS: 113-D3250100-102
[  244.075241] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  244.075243] vfio-pci 0000:86:00.0: AMD_NAVI14: bus reset disabled? yes
[  244.075248] vfio-pci 0000:86:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 55fdd49, mp1 intr enabled? yes, bl ready? yes
[  244.075249] vfio-pci 0000:86:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
[  244.075277] vfio-pci 0000:86:00.0: AMD_NAVI14: begin psp mode 1 reset
[  244.582343] vfio-pci 0000:86:00.0: AMD_NAVI14: mode1 reset succeeded
[  244.582454] vfio-pci 0000:86:00.0: AMD_NAVI14: PSP mode1 reset successful
[  244.582458] vfio-pci 0000:86:00.0: AMD_NAVI14: performing post-reset
[  244.607340] vfio-pci 0000:86:00.0: AMD_NAVI14: reset result = 0`

Start VM again:
`[  322.460703] vfio-pci 0000:86:00.0: enabling device (0400 -> 0403)
[  322.460853] vfio-pci 0000:86:00.0: AMD_NAVI14: version 1.1
[  322.460854] vfio-pci 0000:86:00.0: AMD_NAVI14: performing pre-reset
[  322.472755] vfio-pci 0000:86:00.0: AMD_NAVI14: performing reset
[  322.509116] ATOM BIOS: 113-D3250100-102
[  322.509118] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  322.509121] vfio-pci 0000:86:00.0: AMD_NAVI14: bus reset disabled? yes
[  322.509126] vfio-pci 0000:86:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 55fdd49, mp1 intr enabled? yes, bl ready? yes
[  322.509127] vfio-pci 0000:86:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
[  322.509155] vfio-pci 0000:86:00.0: AMD_NAVI14: begin psp mode 1 reset
[  322.755198] vfio-pci 0000:86:00.0: AMD_NAVI14: timed out waiting for PSP to reach valid state, but continuing anyway
[  323.263760] vfio-pci 0000:86:00.0: AMD_NAVI14: mode1 reset succeeded
[  323.263870] vfio-pci 0000:86:00.0: AMD_NAVI14: PSP mode1 reset successful
[  323.263875] vfio-pci 0000:86:00.0: AMD_NAVI14: performing post-reset
[  323.287716] vfio-pci 0000:86:00.0: AMD_NAVI14: reset result = 0
[  323.287878] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x19@0x270
[  323.287891] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x1b@0x2d0
[  323.287895] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x25@0x400
[  323.287896] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x26@0x410
[  323.287898] vfio_ecap_init: 0000:86:00.0 hiding ecap 0x27@0x440
[  323.318822] vfio-pci 0000:86:00.0: AMD_NAVI14: version 1.1
[  323.318823] vfio-pci 0000:86:00.0: AMD_NAVI14: performing pre-reset
[  323.330749] vfio-pci 0000:86:00.0: AMD_NAVI14: performing reset
[  323.367851] ATOM BIOS: 113-D3250100-102
[  323.367853] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  323.367855] vfio-pci 0000:86:00.0: AMD_NAVI14: bus reset disabled? yes
[  323.367860] vfio-pci 0000:86:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 55fdd49, mp1 intr enabled? yes, bl ready? yes
[  323.367861] vfio-pci 0000:86:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
[  323.367889] vfio-pci 0000:86:00.0: AMD_NAVI14: begin psp mode 1 reset
[  323.614481] vfio-pci 0000:86:00.0: AMD_NAVI14: timed out waiting for PSP to reach valid state, but continuing anyway
[  324.119785] vfio-pci 0000:86:00.0: AMD_NAVI14: mode1 reset succeeded
[  324.119894] vfio-pci 0000:86:00.0: AMD_NAVI14: PSP mode1 reset successful
[  324.119899] vfio-pci 0000:86:00.0: AMD_NAVI14: performing post-reset
[  324.143751] vfio-pci 0000:86:00.0: AMD_NAVI14: reset result = 0`

relm, 2021-07-24:
FWIW, I have a Radeon Pro W5500 and vendor-reset is working for me. Did not try with a Linux guest but I was able to restart successfully in a macOS guest, as well as shut down the VM and start it back up again. 

Not sure if other commits have fixed it since but are you sure the amdgpu driver isn't reattaching to the card between VM restarts? With the way I have things setup only vfio-pci is.

VM start:
```
[   79.032470] vfio-pci 0000:14:00.0: enabling device (0002 -> 0003)
[   79.032658] vfio-pci 0000:14:00.0: AMD_NAVI14: version 1.1
[   79.032660] vfio-pci 0000:14:00.0: AMD_NAVI14: performing pre-reset
[   79.052521] vfio-pci 0000:14:00.0: AMD_NAVI14: performing reset
[   79.092395] ATOM BIOS: 113-D3250100-102
[   79.092396] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[   79.363370] vfio-pci 0000:14:00.0: AMD_NAVI14: bus reset disabled? yes
[   79.363376] vfio-pci 0000:14:00.0: AMD_NAVI14: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[   79.363379] vfio-pci 0000:14:00.0: AMD_NAVI14: performing post-reset
[   79.400467] vfio-pci 0000:14:00.0: AMD_NAVI14: reset result = 0
[   79.400683] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
[   79.400700] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
[   79.400705] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x25@0x400
[   79.400707] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x26@0x410
[   79.400709] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x27@0x440
[   79.420474] vfio-pci 0000:14:00.1: enabling device (0000 -> 0002)
[   80.656976] vfio-pci 0000:14:00.0: AMD_NAVI14: version 1.1
[   80.656981] vfio-pci 0000:14:00.0: AMD_NAVI14: performing pre-reset
[   80.657165] vfio-pci 0000:14:00.0: AMD_NAVI14: performing reset
[   80.693326] ATOM BIOS: 113-D3250100-102
[   80.693328] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[   80.732510] virbr0: port 1(vnet0) entered forwarding state
[   80.732515] virbr0: topology change detected, propagating
[   80.933674] vfio-pci 0000:14:00.0: AMD_NAVI14: bus reset disabled? yes
[   80.933679] vfio-pci 0000:14:00.0: AMD_NAVI14: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[   80.933681] vfio-pci 0000:14:00.0: AMD_NAVI14: performing post-reset
[   80.972499] vfio-pci 0000:14:00.0: AMD_NAVI14: reset result = 0
```

Reboot:
```
[  162.178122] vfio-pci 0000:14:00.0: AMD_NAVI14: version 1.1
[  162.178127] vfio-pci 0000:14:00.0: AMD_NAVI14: performing pre-reset
[  162.178301] vfio-pci 0000:14:00.0: AMD_NAVI14: performing reset
[  162.218016] ATOM BIOS: 113-D3250100-102
[  162.218018] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  162.218020] vfio-pci 0000:14:00.0: AMD_NAVI14: bus reset disabled? yes
[  162.218025] vfio-pci 0000:14:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 691b826, mp1 intr enabled? yes, bl ready? yes
[  162.218027] vfio-pci 0000:14:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
[  162.218124] vfio-pci 0000:14:00.0: AMD_NAVI14: begin psp mode 1 reset
[  162.748828] vfio-pci 0000:14:00.0: AMD_NAVI14: mode1 reset succeeded
[  163.828834] vfio-pci 0000:14:00.0: AMD_NAVI14: PSP mode1 reset successful
[  163.828843] vfio-pci 0000:14:00.0: AMD_NAVI14: performing post-reset
[  163.868840] vfio-pci 0000:14:00.0: AMD_NAVI14: reset result = 0
```

Shutdown and start:
```
[  262.756953] vfio-pci 0000:14:00.0: AMD_NAVI14: version 1.1
[  262.756958] vfio-pci 0000:14:00.0: AMD_NAVI14: performing pre-reset
[  262.776744] vfio-pci 0000:14:00.0: AMD_NAVI14: performing reset
[  262.816682] ATOM BIOS: 113-D3250100-102
[  262.816685] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  262.816688] vfio-pci 0000:14:00.0: AMD_NAVI14: bus reset disabled? yes
[  262.816693] vfio-pci 0000:14:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: acc0e87, mp1 intr enabled? yes, bl ready? yes
[  262.816695] vfio-pci 0000:14:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
[  262.816794] vfio-pci 0000:14:00.0: AMD_NAVI14: begin psp mode 1 reset
[  263.324678] vfio-pci 0000:14:00.0: AMD_NAVI14: mode1 reset succeeded
[  264.512475] vfio-pci 0000:14:00.0: AMD_NAVI14: PSP mode1 reset successful
[  264.512485] vfio-pci 0000:14:00.0: AMD_NAVI14: performing post-reset
[  264.552696] vfio-pci 0000:14:00.0: AMD_NAVI14: reset result = 0
[  309.132221] vfio-pci 0000:14:00.0: AMD_NAVI14: bus reset disabled? yes
[  309.132227] vfio-pci 0000:14:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  309.132229] vfio-pci 0000:14:00.0: AMD_NAVI14: performing post-reset
[  309.168757] vfio-pci 0000:14:00.0: AMD_NAVI14: reset result = 0
[  309.168973] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
[  309.168989] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
[  309.168995] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x25@0x400
[  309.168997] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x26@0x410
[  309.168999] vfio-pci 0000:14:00.0: vfio_ecap_init: hiding ecap 0x27@0x440
[  310.433027] vfio-pci 0000:14:00.0: AMD_NAVI14: version 1.1
[  310.433031] vfio-pci 0000:14:00.0: AMD_NAVI14: performing pre-reset
[  310.433249] vfio-pci 0000:14:00.0: AMD_NAVI14: performing reset
[  310.473202] ATOM BIOS: 113-D3250100-102
[  310.473204] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  310.492774] virbr0: port 1(vnet1) entered forwarding state
[  310.492781] virbr0: topology change detected, propagating
[  310.745106] vfio-pci 0000:14:00.0: AMD_NAVI14: bus reset disabled? yes
[  310.745112] vfio-pci 0000:14:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  310.745115] vfio-pci 0000:14:00.0: AMD_NAVI14: performing post-reset
[  310.784836] vfio-pci 0000:14:00.0: AMD_NAVI14: reset result = 0
```

thielemans, 2021-07-25:
Thanks for letting me know it is not a hardware specific issue. I will migrate away from CentOS 8 anyway as it is going EOL soon, so it would be a good time to revisit this issue in the coming weeks.

gnif, 2021-08-04:
As it stands, there is nothing we can do for you if the reset is not working. We have no new information and there is nothing more we can try. If this was stable and worked for everyone we would try to have these fixes upstreamed into the mainline kernel.

closed 2021-08-04 by gnif