gnif/vendor-reset #25: SAPPHIRE Pulse Radeon RX 5500 XT SF 4G reset does not work

msmuenchen, 2021-02-28:
Hi,

something is iffy with the 5500 XT - I'm getting a similar behavior as the person in #23.

My setup:
* Machine: Mac Pro 4.1 (flashed to 5.1), 2x Xeon E5520, 32GB RAM
* OS Debian Testing, kernel 5.10.0-3-amd64 (both in host and guest)
* vendor-reset master, commit 225a49a40941e350899e456366265cf82b87ad25
* Primary GPU (slot 1): RX 5500 XT - amdgpu blacklisted, vfio-pci stub set up
* Secondary GPU NVIDIA GF 710 - used for host graphics
* QEMU-KVM machine Q35 in UEFI mode, OVMF 2020.11-2

Host dmesg after first start of VM:
```
[  580.162288] vfio-pci 0000:0c:00.0: enabling device (0002 -> 0003)
[  580.162716] vfio-pci 0000:0c:00.0: AMD_NAVI14: version 1.1
[  580.162966] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing pre-reset
[  580.182421] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing reset
[  580.218019] ATOM BIOS: 113-2E4275U-OIM
[  580.218290] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  580.401033] vfio-pci 0000:0c:00.0: AMD_NAVI14: bus reset disabled? yes
[  580.401262] vfio-pci 0000:0c:00.0: AMD_NAVI14: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  580.401824] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing post-reset
[  580.442315] vfio-pci 0000:0c:00.0: AMD_NAVI14: reset result = 0
[  580.442733] vfio-pci 0000:0c:00.0: vfio_ecap_init: hiding ecap 0x19@0x270
[  580.443139] vfio-pci 0000:0c:00.0: vfio_ecap_init: hiding ecap 0x1b@0x2d0
[  580.443779] vfio-pci 0000:0c:00.0: vfio_ecap_init: hiding ecap 0x25@0x400
[  580.444425] vfio-pci 0000:0c:00.0: vfio_ecap_init: hiding ecap 0x26@0x410
[  580.445190] vfio-pci 0000:0c:00.0: vfio_ecap_init: hiding ecap 0x27@0x440
[  580.898393] vfio-pci 0000:00:1d.7: vfio_cap_init: hiding cap 0xa@0x58
[  581.466363] vfio-pci 0000:0c:00.0: AMD_NAVI14: version 1.1
[  581.466700] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing pre-reset
[  581.486386] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing reset
[  581.522097] ATOM BIOS: 113-2E4275U-OIM
[  581.522430] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  581.705222] vfio-pci 0000:0c:00.0: AMD_NAVI14: bus reset disabled? yes
[  581.705452] vfio-pci 0000:0c:00.0: AMD_NAVI14: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? yes
[  581.705880] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing post-reset
[  581.746278] vfio-pci 0000:0c:00.0: AMD_NAVI14: reset result = 0
```

Host dmesg after shutdown triggered in VM:
```
[  684.065413] vfio-pci 0000:0c:00.0: AMD_NAVI14: version 1.1
[  684.065760] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing pre-reset
[  684.084923] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing reset
[  684.121323] ATOM BIOS: 113-2E4275U-OIM
[  684.121645] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  684.121961] vfio-pci 0000:0c:00.0: AMD_NAVI14: bus reset disabled? yes
[  684.122432] vfio-pci 0000:0c:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 23976f79, mp1 intr enabled? yes, bl ready? yes
[  684.123055] vfio-pci 0000:0c:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
[  684.123600] vfio-pci 0000:0c:00.0: SMU error 0xfd
[  684.124158] vfio-pci 0000:0c:00.0: Failed to send message 0x2a: return 0xfd
[  684.124865] vfio-pci 0000:0c:00.0: SMU error 0xfd
[  684.125347] vfio-pci 0000:0c:00.0: AMD_NAVI14: begin psp mode 1 reset
[  684.652771] vfio-pci 0000:0c:00.0: AMD_NAVI14: mode1 reset succeeded
[  684.653184] vfio-pci 0000:0c:00.0: AMD_NAVI14: PSP mode1 reset successful
[  684.653505] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing post-reset
[  684.692830] vfio-pci 0000:0c:00.0: AMD_NAVI14: reset result = 0
```

Host dmesg after second time starting of VM:
```
[  754.219816] vfio-pci 0000:0c:00.0: enabling device (0400 -> 0403)
[  754.220256] vfio-pci 0000:0c:00.0: AMD_NAVI14: version 1.1
[  754.220740] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing pre-reset
[  754.239801] vfio-pci 0000:0c:00.0: AMD_NAVI14: performing reset
[  754.276159] ATOM BIOS: 113-2E4275U-OIM
[  754.276405] vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
[  754.276978] vfio-pci 0000:0c:00.0: AMD_NAVI14: bus reset disabled? yes
[... 106 lines cut ...]
[  102.375088]  ? device_driver_attach+0xb0/0xb0
[  102.375703]  bus_for_each_dev+0x78/0xc0
[  102.376238]  bus_add_driver+0x12b/0x1e0
[  102.376774]  driver_register+0x8b/0xe0
[  102.377320]  ? 0xffffffffc066b000
[  102.377795]  do_one_initcall+0x44/0x1d0
[  102.378331]  ? do_init_module+0x23/0x260
[  102.378882]  ? kmem_cache_alloc_trace+0xf5/0x200
[  102.379564]  do_init_module+0x5c/0x260
[  102.380103]  __do_sys_finit_module+0xb1/0x110
[  102.380712]  do_syscall_64+0x33/0x80
[  102.381224]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  102.382042] RIP: 0033:0x7fe80e7249b9
[  102.382568] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a7 54 0c 00 f7 d8 6
[  102.385147] RSP: 002b:00007fff230ece08 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  102.386254] RAX: ffffffffffffffda RBX: 0000557d574f2c70 RCX: 00007fe80e7249b9
[  102.387255] RDX: 0000000000000000 RSI: 0000557d55782260 RDI: 0000000000000003
[  102.388268] RBP: 0000000000040000 R08: 0000000000000000 R09: 0000557d574f2e30
[  102.389271] R10: 0000000000000003 R11: 0000000000000246 R12: 0000557d55782260
[  102.390271] R13: 0000000000000000 R14: 0000557d574f2d80 R15: 0000557d574f2c70
[  102.391383] [drm:amdgpu_atombios_init [amdgpu]] *ERROR* Failed to create device file for VBIOS version
[  102.392666] amdgpu 0000:06:00.0: amdgpu: amdgpu_atombios_init failed
[  102.397325] amdgpu 0000:06:00.0: amdgpu: Fatal error during GPU init
[  102.398982] amdgpu: probe of 0000:06:00.0 failed with error -17
```

lnxd, 2021-03-15:
I'm getting the same with an Asus ROG Strix Radeon RX 5500 XT 8GB card, Unraid as the host.

gnif, 2021-03-15:
`amdgpu` does not release the card in a sane way, we can not reset from this. vendor-reset is to allow you to reboot your guest, not to unbind/rebind the gpu on the host.

closed 2021-03-15 by gnif

lnxd, 2021-03-16:
Thank you. I wonder if that's what's happening in my case as well then.

Just one thing, 5500 XT is listed as supported on the readme, am I correct in assuming that this is either not tested or not supported with all 5500 XTs? In which case, can this please be updated to indicate that it's not fully supported? Just as I cross referenced the list before purchase due to the success I had with your work and my RX 580 and I'd hate for that to happen to someone else. 

Amazing job though, this module was lifesaving for my RX 580, so I'm hoping there's some way I can get the 5500 XT working as well.

Apologies in advance if this comment re-opens the issue.