gnif/vendor-reset #26: Compilation Error

GeorgeGhost, 2021-03-11:
Tried to `dkms install . ` on linux-lqx 5.11.5 (Manjaro), could not compile:

```
Creating symlink /var/lib/dkms/vendor-reset/0.1.1/source ->
                 /usr/src/vendor-reset-0.1.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.11.5-lqx1-1-lqx KDIR=/usr/lib/modules/5.11.5-lqx1-1-lqx/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.11.5-lqx1-1-lqx (x86_64)
Consult /var/lib/dkms/vendor-reset/0.1.1/build/make.log for more information.
```
/var/lib/dkms/vendor-reset/0.1.1/build/make.log
```
DKMS make.log for vendor-reset-0.1.1 for kernel 5.11.5-lqx1-1-lqx (x86_64)
Чт 11 мар 2021 09:20:51 GMT
make -C /usr/lib/modules/5.11.5-lqx1-1-lqx/build M=/var/lib/dkms/vendor-reset/0.1.1/build modules
make[1]: Entering directory '/usr/lib/modules/5.11.5-lqx1-1-lqx/build'
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/module.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/vendor-reset-dev.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/hook.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/common.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/compat.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/firmware.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/navi10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/polaris10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/vega10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/vega20.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_device.o
In file included from /var/lib/dkms/vendor-reset/0.1.1/build/src/hook.c:21:
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.h:33:21: error: field ‘ops’ has incomplete type
   33 |   struct ftrace_ops ops;
      |                     ^~~
In file included from /var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:27:
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.h:33:21: error: field ‘ops’ has incomplete type
   33 |   struct ftrace_ops ops;
      |                     ^~~
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:56:110: warning: ‘struct ftrace_regs’ declared inside parameter list will not be visible outside of this definition or declaration
   56 | static void notrace fh_trace_thunk(unsigned long ip, unsigned long parent_ip, struct ftrace_ops *ops, struct ftrace_regs *fregs)
      |                                                                                                              ^~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/vendor-reset/0.1.1/build/src/hook.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c: In function ‘fh_trace_thunk’:
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:59:10: error: implicit declaration of function ‘ftrace_get_regs’ [-Werror=implicit-function-declaration]
   59 |   regs = ftrace_get_regs(fregs);
      |          ^~~~~~~~~~~~~~~
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:59:8: warning: assignment to ‘struct pt_regs *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
   59 |   regs = ftrace_get_regs(fregs);
      |        ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:61:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
   61 |   struct ftrace_hook *hook = to_ftrace_hook(ops);
      |   ^~~~~~
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c: In function ‘fh_install_hook’:
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:76:21: error: ‘FTRACE_OPS_FL_SAVE_REGS’ undeclared (first use in this function)
   76 |   hook->ops.flags = FTRACE_OPS_FL_SAVE_REGS |
      |                     ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:76:21: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:77:21: error: ‘FTRACE_OPS_FL_IPMODIFY’ undeclared (first use in this function)
   77 |                     FTRACE_OPS_FL_IPMODIFY;
      |                     ^~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.o] Error 1
make[1]: *** [Makefile:1800: /var/lib/dkms/vendor-reset/0.1.1/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.11.5-lqx1-1-lqx/build'
make: *** [Makefile:8: build] Error 2
```

amadejkastelic, 2021-03-20:
Same issue for 5.11.6-xanmod1-cacule.

amadejkastelic, 2021-03-20:
Just figured it out. You need to have these kernel options: https://github.com/gnif/vendor-reset#requirements

GeorgeGhost, 2021-03-20:
> Just figured it out. You need to have these kernel options: https://github.com/gnif/vendor-reset#requirements. I just edited the PKGBUILD and rebuilt the kernel.

Hmm, I have double checked my current config options using `zcat /proc/config.gz | grep -e CONFIG_FTRACE -e CONFIG_KPROBES -e CONFIG_PCI_QUIRKS -e CONFIG_KALLSYMS -e CONFIG_KALLSYMS_ALL -e CONFIG_FUNCTION_TRACER -e Linux/x86`
And got following output:
```
# Linux/x86 5.11.7-lqx1 Kernel Configuration
CONFIG_KALLSYMS=y
CONFIG_KALLSYMS_ALL=y
CONFIG_KALLSYMS_ABSOLUTE_PERCPU=y
CONFIG_KALLSYMS_BASE_RELATIVE=y
CONFIG_KPROBES=y
CONFIG_PCI_QUIRKS=y
CONFIG_FTRACE=y
```
So I believe that all of the required options are enabled, I do have linux-headers installed for that kernel

philon123, 2021-08-03:
I have a similar error when compiling on Proxmox VE 5.4 - The cmd shared by @GeorgeGhost didn't work for diagnosing kernel args, because I didn't have a `/proc/config.gz`, but I do have a `/boot/config-4.15.18-30-pve`, which includes all the required kernel args. So, I think the module should compile fine!

However, here is the detailed error log:
`DKMS make.log for vendor-reset-0.1.1 for kernel 4.15.18-30-pve (x86_64)
Tue Aug  3 22:11:30 CEST 2021
make -C /lib/modules/4.15.18-30-pve/build M=/var/lib/dkms/vendor-reset/0.1.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.18-30-pve'
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/module.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/vendor-reset-dev.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/hook.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/common.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/compat.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/firmware.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/navi10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/polaris10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/vega10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/vega20.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_device.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_discovery.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_atomfirmware.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_bios.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/atom.o
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c: In function ‘fh_trace_thunk’:
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:63:8: error: implicit declaration of function ‘within_module’ [-Werror=implicit-function-declaration]
   if (!within_module(parent_ip, THIS_MODULE))
        ^~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/common_baco.o
cc1: some warnings being treated as errors
scripts/Makefile.build:322: recipe for target '/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.o' failed
make[2]: *** [/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/navi10_reg_init.o
Makefile:1582: recipe for target '_module_/var/lib/dkms/vendor-reset/0.1.1/build' failed
make[1]: *** [_module_/var/lib/dkms/vendor-reset/0.1.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.18-30-pve'
Makefile:8: recipe for target 'build' failed
make: *** [build] Error 2`

Maybe my Kernel is too old?

NaN-git, 2021-08-03:
@philon123 Did you try to include `linux/module.h` in ftrace.c yet?

gnif, 2021-08-04:
`4.15.18-30-pve` is too old. There was no issue here other than a failure to follow the configure flag documentation.

closed 2021-08-04 by gnif