gnif/vendor-reset #35: RX 6800

TheCherry, 2021-07-05:
Does that work for the RX 6800 too?

gnif, 2021-07-05:
The rx6800 does not need a reset fix.

closed 2021-07-05 by gnif

labor4, 2021-07-05:
> The rx6800 does not need a reset fix.

i was wondering this about the whole new 6000 series, but could only guess from the lack of negative claims. in times of shortage this is an uncertain conclusion to make. 

is there a reset-bug-reported matrix for gpus somewhere? if not, would this be a good idea?

gnif, 2021-07-05:
The entire 6000 series are fine, this only affects Vega10, Vega20, and Navi10 GPUs.

bobafetthotmail, 2021-07-06:
@labor4 You can follow **level1techs** on youtube (or in the forums). Wendell is very involved in the KVM/passthrough/Linux scene, he did testing and reports on 6000 series cards a few months ago.

TheCherry, 2021-07-15:
> The rx6800 does not need a reset fix.

This is not true. At the moment I have to start always a bash script or reboot the host, after I shutdown my Windows VM.
Else the VM can't attach the GPU and boot.

At the moment I use this script:
` 
echo "1" | tee -a /sys/bus/pci/devices/0000\:2d\:00.1/remove
sleep 1
echo -n mem > /sys/power/state
sleep 1
echo "1" | tee -a /sys/bus/pci/rescan
`
But I have often to start it 2-3 times.

gnif, 2021-07-15:
This is an issue with your system then, rx6800 does not need any patches to allow it to restart, if it did even your workaround would not work as the result of a failed reset is a completely dead cart that requires a full reboot to restore it.

flamme-demon, 2021-07-22:
hi, 

For my part I am with a 6700 on xpc-ng and I confirm that after reboot there is an error 43

gnif, 2021-07-23:
> error 43

This is NOT the reset bug, the reset bug crashes the card out completely, making it fall off the PCIe BUS, breaking your system entirely.

flamme-demon, 2021-07-23:
After shutting down or restarting the VM, the device is not usable at all until the physical machine is fully restarted or it is located

gnif, 2021-07-23:
> Windows displays it as error 43

You can clearly still start a VM up with vfio-pci bound to the device. I say again, this is NOT the reset bug, you have something else wrong with your system, perhaps a motherboard that isn't playing nice with VFIO/IOMMU, as is common.