gnif/vendor-reset #36: Feature Request: Adding Support to [AMD/ATI] Polaris 22 XT [Radeon RX Vega M GH]

AmandaBeuno, 2021-07-29:
This is really a feature request, not an issue reporting. 

Would it be possible to add support to the [AMD/ATI] Polaris 22 XT [Radeon RX Vega M GH]. 

The gpu id is 1002:694c
the audio counterpart is 1002:ab08

Please let me know if there is any more data needed and I will be willing to provide them.

All the best,

gnif, 2021-07-29:
It's not a simple matter of just adding support. vendor-reset either works for you or it doesn't. These reset sequences are specific to the models listed and are only partially understood at best.

closed 2021-07-29 by gnif

AmandaBeuno, 2021-07-29:
@gnif . Thank you for the quick response and of course for the great project :)

I totally understand that, I just thought that the reset sequences wouldn't affect the gpu as its id is not there in the list. I have to admit that it is quite beyond me at this stage. Thanks again @gnif