gnif/vendor-reset #37: compile on arm64 server, kernel version linux-4.19

xieming1988, 2021-08-04:
/root/xm/kernel/vendor-reset-master/src/ftrace.c:63:11: 错误：‘struct pt_regs’ has no member named ‘ip’; did you mean ‘sp’?
     regs->ip = (unsigned long)hook->function;
           ^~
           sp
then, I changed 'regs->ip' to 'regs->sp'， compiled successd.
but when I insmod vendor-reset.ko， dmesg report error:
 ftrace_set_filter_ip() failed: -22

gnif, 2021-08-04:
Old unsupported kernel, why raise an issue?

xieming1988, 2021-08-05:
(⊙o⊙)… on arm64 server， ’struct pt_regs’ has no member nameniothat is not kernel version reason

gnif, 2021-08-05:
It 100% is a kernel version reason, you're using a kernel that doesn't have the struct members named as expected.

Nable80, 2021-08-05:
Instruction pointer is called ``pc`` (program counter) on ARM, not ``ip``. And stack pointer (``sp``) is a completely different register.

gnif, 2021-08-05:
It's still being build for arm64 instead of amd64, as such it's a different version and we do not support it.

Nable80, 2021-08-05:
arm64's instruction pointer is called ``pc`` too. Is it bad if someone with access to ARM/MIPS/E2K/etc hardware adds support for it? Would it hurt anyhow? Yes, it's possible to use PCI(-E) GPUs on non-x86 hardware.

gnif, 2021-08-05:
A PR to fix this properly would be accepted, however as none of these resets are tested on arm64 and may not even be safe to do so as we are poking at registers we do not fully understand.

xieming1988, 2021-08-05:
> arm64's instruction pointer is called `pc` too. Is it bad if someone with access to ARM/MIPS/E2K/etc hardware adds support for it? Would it hurt anyhow? Yes, it's possible to use PCI(-E) GPUs on non-x86 hardware.

(⊙o⊙)…Did you solve this problem on arm64?

Nable80, 2021-08-05:
> Did you solve this problem on arm64?

I just tried to clarify the misunderstanding that I noticed here. If you got my idea, you can check it on your hardware or provide access to it somehow. I have no arm64 hardware now.