gnif/vendor-reset #40: Problems on 21.04

mrweiner, 2021-09-03:
This was working for me on popos 20.04, but I've been getting reset issues since upgrading to 21.04. Running a 5700XT fwiw.

dev-sda1, 2021-09-05:
have you tried reinstalling the module?

mrweiner, 2021-09-09:
Yeah I've uninstalled+removed and then reinstalled with dkms

Rydako, 2021-09-12:
Can confirm I have the same issue on 21.04. Have swapped to the system suspect method to get my gpu passthrough working again.

mrweiner, 2021-09-13:
What's the system suspend method?

mwpastore, 2021-09-15:
Have you tried not passing the audio function per #29?

mrweiner, 2021-09-15:
@mwpastore no I hadn't, but removing audio did actually fix my issue. Thanks! Not sure if we should leave this open in case @Rydako is seeing something different?

mrweiner, 2021-09-17:
Actually it seems...kind of fixed. Seems like it's hit or miss whether or not I'm able to actually get any graphics out of the passthrough card even though virt manager itself isn't throwing any nasty errors.

Rydako, 2021-09-27:
> @Rydako What's the system suspend method?

`#!/bin/bash

# This script must be run immediately after you shut down the VM. It doesn't work if the GPU has been left for too long
# following a shut down. It doesn't work if the VM is rebooted.

# Remove/Power off the Vega GPU like uninstalling devices in Windows device manager
echo “1” | sudo tee -a /sys/bus/pci/devices/0000:44:00.0/remove
echo “1” | sudo tee -a /sys/bus/pci/devices/0000:44:00.1/remove

# Suspend to RAM
systemctl suspend

# Read any user input
read input

# Change permisions of the PCI rescan command
sudo chmod 777 /sys/bus/pci/rescan

# Rescan PCI devices to reinitialise the vega GPU
sudo echo 1 > /sys/bus/pci/rescan`

> @mwpastore no I hadn't, but removing audio did actually fix my issue. Thanks! Not sure if we should leave this open in case @Rydako is seeing something different?

I've actually gone and accidentally deleted my VM, lol. Go ahead and close this - if you're not having issues, this one is likely on my end as well. I need audio output, so can't remove the audio device.. Not sure when I'll have time to play around again though, no time at the moment, unfortunately.

dev-sda1, 2021-09-27:
Something i've stumbled upon recently that might help with things, as Ubuntu recently made a full switch to Pipewire which apparently doesn't like it when you start vms sometimes:

Run this before starting your VM and see if there's any difference? ```systemctl --user stop pipewire.service pipewire.socket```