gnif/vendor-reset #44: Black Screen after apply vendor reset

spcweb, 2021-11-22:
Hi, 

i have a working VM with QUEMU/KVM on linux arch based distro (i'm using Garuda). 

I have made a passthrough of my RX580. The VM working well, but when i shutdown the host doesn't apper back. 

So i have install vendor-reset, follow the instructions. But, after the install the virtual machine doesn't boot anymore, and the screen remains black. 

Someone have some tips about that? 

Thanks!

fire-burner, 2021-11-30:
Which kernel are you running? I am having issues since 5.15 too.
Also check if the module is loaded after boot (dmesg | grep vendor, which should show sth. like this "... vendor_reset_hook: installed")

spcweb, 2021-12-01:
My Kernel is: 5.15.5-zen1-1-zen... there is a way to remove the vendor reset?

fire-burner, 2021-12-01:
then it is most probably the same issue as #46