gnif/vendor-reset #45: After returning to host login screen, blackscreen when loggin in.

Fisk24, 2021-11-24:
So after having installed vendor reset I can say that it did work to get me back to the gnome login screen. However when I try to actually login, the screen flickers a few times and I'm stuck at a black screen and the desktop never comes back. To be clear, this doesn't happen after a full reboot. Only after I shutdown my VM.

Here's a concise series of events:
- Boot host machine and login as normal with a working desktop.
- Run my win10 VM with gpu passthrough
- Shutdown VM
- Back at host login screen now
- Try to login
- Blackscreen... :(

I have a Sapphire RX5700xt if that helps

fire-burner, 2021-11-30:
Which kernel are you running? I am having issues since 5.15 too.
Also check if the module is loaded after boot (dmesg | grep vendor, which should show sth. like this "... vendor_reset_hook: installed")

Fisk24, 2021-12-02:
Using Kernel 5.10 LTS Manjaro. And I have already confirmed that the module is loaded using the method you mentioned. Unfortunately since posting this bug I managed to break my VM beyond repair and I dont have time to set it up again right now.
Will try again in a few days hopefully.