gnif/vendor-reset #53: 6800xt

Theempireshrikesback, 2022-02-06:
Hello,

I am newer to Linux and have access to a 6800xt and a 5950x. I didn't see the 6800xt listed as supported and was wondering if I could be of any assistance in adding that device to the supported list. FWIW, I haven't looked far enough into this to even be certain that this piece of silicon still has that issue, but based on AMD's history I'm assuming it is also impacted.

Cheers

LukeShortCloud, 2022-04-05:
@Theempireshrikesback Good news for you: the AMD Radeon RX 6800 XT was one of the first AMD dGPUs that does not require the `vendor-reset` workaround!

https://www.youtube.com/watch?v=ykiU49gTNak

nergdron, 2022-07-08:
so, I've got an XFX 6800XT, and I'm definitely seeing this behaviour with it, I still have to reboot my entire PC after shutting down my windows VM before start the VM again. this is arch, with Linux kernel version 5.15.52 (at last test), so it does seem to be an issue with some 6xxx series cards.

TimGels, 2022-11-25:
Same problem on my end with the AMD reference RX 6900 XT, also running Arch. Kernel version 6.0.9.