gnif/vendor-reset #59: Is it possible to support iGPU like 5900HX?

Pandaaaa906, 2022-06-08:


zncs520, 2023-02-03:
我是刚刚入手的6900HX小主机 ，已经成功直通输出到显示器，我的方案是改QUMU的启动BIOS，附加VBIOS启动
结果很成功。不打驱动VM显示输出很正常，重启关机都不印象输出

驱动后就会有这个RESET BUG。

楼主至今没有解决重启BUG吗？

附上我的核心文件：

[6900HX一键直通核显.zip](https://github.com/gnif/vendor-reset/files/10576587/6900HX.zip)

WSt89, 2023-03-08:
I also need support for a Vega 8 iGPU (Ryzen 7 PRO 5850U with Radeon Graphics). I tried to load vendor-reset module for my proxmox system but without success.

DocMAX, 2023-04-15:
Sadly not supported generation of CPU. And @gnif is not interested in fixing that. :-( So we have to wait for AMD to fix their bugs.

cRaZy-bisCuiTs, 2023-08-19:
> Sadly not supported generation of CPU. And @gnif is not interested in fixing that. :-( So we have to wait for AMD to fix their bugs.

Is there an open bug report? Where do we need to report that? amdgpu developing bug tracker [0]?

[0]
https://gitlab.freedesktop.org/drm/amd/-/issues

WSt89, 2023-08-19:
I moved my GPU dependent services from VMs to LXC and I mounted the GPU from host system into the containers. It works without any issues and it consumes even less resources.

forthal, 2024-10-04:
> I moved my GPU dependent services from VMs to LXC and I mounted the GPU from host system into the containers. It works without any issues and it even consumes less resources.

I tried the same, but ended up with problem because I need Windows to launch Windows apps with GPU :(

DocMAX, 2024-10-04:
> I moved my GPU dependent services from VMs to LXC and I mounted the GPU from host system into the containers. It works without any issues and it even consumes less resources.

Same here and it works surprisingly well

ballerburg9005, 2024-12-06:
I made a PR to improve the README on how to use vendor-reset with unsupported GPUs.

https://github.com/gnif/vendor-reset/pull/89

I was deceived by first impressions, that it wouldn't be as simple as adding the product ID to the list, for the iGPU of Ryzen 5 5600G (Cezanne / Vega 7 / CHIP_RENOIR). However it was as simple as adding the product ID to the list.

Specifically for iGPUs you should try reset strategy NAVI10, and my guess is that the other strategies won't work. But since there are only four, it is worth a shot.