gnif/vendor-reset #63: Request to add the hardware id of the amd APU

wxzmz, 2022-09-03:
There is also a reset error when amd apu passthrough
Can add hardware ID

Similar to the following
https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c

lines 1789
	/* carrizo */
	{0x1002, 0x9870, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_CARRIZO|AMD_IS_APU},
	{0x1002, 0x9874, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_CARRIZO|AMD_IS_APU},
	{0x1002, 0x9875, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_CARRIZO|AMD_IS_APU},
	{0x1002, 0x9876, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_CARRIZO|AMD_IS_APU},
	{0x1002, 0x9877, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_CARRIZO|AMD_IS_APU},
	/* stoney */
	{0x1002, 0x98E4, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_STONEY|AMD_IS_APU},

lines 1864
	/* Raven */
	{0x1002, 0x15dd, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RAVEN|AMD_IS_APU},
	{0x1002, 0x15d8, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RAVEN|AMD_IS_APU},

lines 1887
	/* Renoir */
	{0x1002, 0x15E7, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},
	{0x1002, 0x1636, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},
	{0x1002, 0x1638, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},
	{0x1002, 0x164C, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},

lines 1915
	/* Yellow Carp */
	{0x1002, 0x164D, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_YELLOW_CARP|AMD_IS_APU},
	{0x1002, 0x1681, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_YELLOW_CARP|AMD_IS_APU},

andywoooo, 2022-09-07:
Yes, I need above too.

thxwork, 2022-09-07:
Yes, I need above too.+1

7immy11, 2022-09-14:
Yes, I need above too.+2

empty99, 2022-09-15:
Yes, I need above too.+4

aersile, 2022-09-15:
Yes, I need above too.+5

jackiegk, 2022-09-15:
Yes, I need above too.+6

agenttaozi, 2022-09-15:
Yes, I need above too.+7

gongzai0, 2022-09-15:
Yes, I need above too.

Jsi210-GitHub, 2022-09-20:
Ya, I need above too.

thisfalse, 2022-09-20:
俺也一样

jieen1, 2022-09-20:
I need above too.

yaj6835669, 2022-09-29:
Yes, I need above too.+8

ThatYears, 2022-10-04:
Yes, I need above too. +13

cppmonkey, 2022-10-06:
Has anyone here managed to pass one through successfully?

I bought the Asrock X570U4D and a Ryzen 7 5700G in the hope that I'd be able to run ProxMox on the hardware and pass the iGPU through to a VM.
Leaving the ASPEED GPU for the PVE console.

Unfortunately the IOMMU grouping had the SATA controllers and the iGPU listed and the system would lockup as soon as the VM was started.
Looks like the IOMMU grouping may have changed with the latest beta BIOS

andywoooo, 2022-10-21:
> 
I succeeded in achieving

andywoooo, 2022-10-21:
> 在座有没有人成功地通过一个？
> 
> 我购买了华擎 X570U4D 和锐龙 7 5700G，希望能够在硬件上运行 ProxMox 并将 iGPU 传递到虚拟机。将高速 GPU 留给 PVE 控制台。
> 
> 不幸的是，IOMMU 分组列出了 SATA 控制器和 iGPU，系统将在 VM 启动后立即锁定。看起来 IOMMU 分组可能已随最新的测试版 BIOS 而更改

> Has anyone here managed to pass one through successfully?
>
> I bought the Asrock X570U4D and a Ryzen 7 5700G in the hope that I'd be able to run ProxMox on the hardware and pass the iGPU through to a [...]

I succeeded in achieving

andywoooo, 2022-10-21:
> Has anyone here managed to pass one through successfully?
>
> I bought the Asrock X570U4D and a Ryzen 7 5700G in the hope that I'd be able to run ProxMox on the hardware and pass the iGPU through to a [...]

I succeeded in achieving

jclendineng, 2022-12-01:
+1 this would be great

Pandaaaa906, 2022-12-17:
Yes, I need above too.+15

ElksInNC, 2022-12-27:
@andywoooo 
> I succeeded in achieving

Can you elaborate?

zncs520, 2023-02-03:
> 是的，我也需要以上。+6900HX
[image]

WSt89, 2023-03-05:
I need passthrough of APU as well.+16