gnif/vendor-reset #73: How to add new GPUs/devices?

cRaZy-bisCuiTs, 2023-08-19:
Does any of you guys understand the code? For me it's a little to close to hardware, so I don't get the implementation.

1. Is it enough to add my IDs to "device-db.h"? or..
2. Do I need to create a new implementation for a new device?

In my case I'd like to reset a Cezanne iGPU.

ballerburg9005, 2024-12-04:
I don't think there is a sane and feasible way how you can add new GPUs without either detailed documentation and hardware specs from AMD, or some of their engineers outlining the procedure to you. I don't have much experience with trying to get such information from manufacturers, but in my imagination this is not something that's very realistic to expect, and whatever stuff you see only was allowed to exist through the workings of pure chance and other magic.

ballerburg9005, 2024-12-06:
https://github.com/gnif/vendor-reset/pull/89

omv-hlab, 2025-02-10:
@ballerburg9005 
Thanks a ton for this information!!!

I have some improvements as it was not completely clear to me what I needed to do to make that hook running:

everything should be done as root

1. add vendor id in my case 1636 for Vega6/Ryzen 3 4350G to `vendor-reset/src/device-db.h`; That ID can be found by `lspci -nnk | grep VGA`
should look like that:
`{PCI_VENDOR_ID_ATI, 0x1636, op, DEVICE_INFO(AMD_NAVI14)}, \`
2. go to vendor-reset and run `make`
3. install vendor_reset module with `dkms install .`
4. echo "vendor-reset" >> /etc/modules
5. to test if the vendor_reset is there you can also do rmmod vendor_reset and lsmod vendor_reset
6. reboot
7. copy vendor-reset/udev/99-vendor-reset.rules to /etc/udev/rules.d/99-vendor-reset.rules and add vendor id here as well: shoulld look like that: ```
ACTION=="add", SUBSYSTEM=="pci", ATTR{vendor}=="0x1002", ATTR{device}=="0x1636", RUN+="/bin/sh -c '/sbin/modprobe vendor-reset; echo device_specific > /sys$env{DEVPATH}/reset_method'"
```
7. run `udevadm control --reload-rules && udevadm trigger`

now `journalctl -xe` should show sth like:

```
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: version 1.1
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: performing pre-reset
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: performing reset
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
Feb 10 01:26:36 omv kernel: ATOM BIOS: 113-RENOIR-035
Feb 10 01:26:36 omv kernel: vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: bus reset disabled? yes
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 2ba060, mp1 intr enabled? yes, bl ready? yes
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: begin psp mode 1 reset
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: mode1 reset succeeded
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: PSP mode1 reset successful
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: performing post-reset
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: reset result = 0
```
which means the hook is working

if not then you most likely have a typo in the vendor id :D (was the case for me) 

BUT I had no luck with NAVI so far. Still Code 43 in Windows and no output in Fedora.
Will test the other methodes as well. Lets see if I will get some output. Hopefully.

Funny enough when I use Windows alone reboots and shutdowns work fine when I use a enable and disable script.

But I want Mac and Linux VMs to use my iGPU as well :)
So hopefully I get a hook like that running.

Dont like the other hooks with bringing the host in sleep mode...

omv-hlab, 2025-02-10:
Though there are basically only vega10, vega20, navi10 and polaris10
Why not Navi14? That exists as well.
If its the same as navi10 how do you know that.

Tnakes

dmuiX, 2025-02-10:
@ballerburg9005 
Thanks a ton for this information!!!

I have some improvements as it was not completely clear to me what I needed to do to make that hook running:

everything should be done as root

1. add vendor id in my case 1636 for Vega6/Ryzen 3 4350G to `vendor-reset/src/device-db.h`; That ID can be found by `lspci -nnk | grep VGA`
should look like that:
`{PCI_VENDOR_ID_ATI, 0x1636, op, DEVICE_INFO(AMD_NAVI14)}, \`
2. go to vendor-reset and run `make`
I think this error `Skipping BTF generation for /home/nasadmin/vendor-reset/vendor-reset.ko due to unavailability of vmlinux` can be ignored,. Tried to fix it which made it worse, in the end had more errors, which I didnt know how to fix.
If more errors occur, most likely not root, missing headers! or some weird kernel config. Actually updated my kernel in the process. which was not a good idea. After a restart everything was good again.
4. install vendor_reset module with `dkms install .`
5. `echo "vendor-reset" >> /etc/modules`
6. to test if the vendor_reset is there you can also do rmmod vendor_reset and lsmod vendor_reset
7. reboot
8. copy vendor-reset/udev/99-vendor-reset.rules to /etc/udev/rules.d/99-vendor-reset.rules and add vendor id here as well: should look like that:```
ACTION=="add", SUBSYSTEM=="pci", ATTR{vendor}=="0x1002", ATTR{device}=="0x1636", RUN+="/bin/sh -c '/sbin/modprobe vendor-reset; echo device_specific > /sys$env{DEVPATH}/reset_method'"
```
7. run `udevadm control --reload-rules && udevadm trigger`

now `journalctl -xe` should show sth like:

```
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: version 1.1
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: performing pre-reset
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: performing reset
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
Feb 10 01:26:36 omv kernel: ATOM BIOS: 113-RENOIR-035
Feb 10 01:26:36 omv kernel: vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: bus reset disabled? yes
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: SMU response reg: 1, sol reg: 2ba060, mp1 intr enabled? yes, bl ready? yes
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: Clearing scratch regs 6 and 7
Feb 10 01:26:36 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: begin psp mode 1 reset
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: mode1 reset succeeded
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: PSP mode1 reset successful
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: performing post-reset
Feb 10 01:26:37 omv kernel: vfio-pci 0000:06:00.0: AMD_NAVI14: reset result = 0
```

which means the hook is working

if not then you most likely have a typo in the vendor id :D (was the case for me) 

BUT I had no luck with NAVI so far. Still Code 43 in Windows and no output in Fedora.
Will test the other methodes as well. Lets see if I will get some output. Hopefully.

Funny enough when I use Windows alone reboots and shutdowns work fine when I use a enable and disable script.

But I want Mac and Linux VMs to use my iGPU as well :)
So hopefully I get a hook like that running.

Dont like the other hooks with bringing the host in sleep mode...

dmuiX, 2025-02-10:
@ballerburg9005 
`Though there are basically only vega10, vega20, navi10 and polaris10`

Why not Navi14? That exists as well.
If its the same as navi10 how do you know that?

Thanks

dmuiX, 2025-09-15:
actually the rom of the hdmi audio device was the problem
when I imported the correct one my error was gone and I had a working iGPU Passthrough with reset, which is amazing!!
Shutdown works just great. 
If I need to reset the vm its breaking. But this is expected.
Havent tested putting the gpu on another vm.

FlakyBlueJay, 2026-02-16:
I presume this won't work for 7000-series Ryzen (Raphael)? AFAIK they're based on Big Navi and the only thing I've seen is to use RadeonResetBugFixService which I'm not sure I want to do on top of Looking Glass as well...

gnif, 2026-05-31:
This is extremely dangerous to do. The IP discovery process and different register mappings between GPU families mean you end up poking at registers that are unrelated to the reset vectors, which can destroy your device. Programming the wrong register with random values can cause permanent damage to the device.

Unless you lift the register defines and reset methods from the kernels `amdgpu` (and any model specific workarounds), you should absolutely not just blindly add extra GPUs to the device database.