gnif/vendor-reset #74: Request support for Ryzen 7 iGPU like 5000U, 5825U

davidngrc, 2023-10-22:
I have one of those small PC, model: MoreFine S500+ , it has an onboard 5825U.
I installed this project, but seems not supported.
please let me know what kind of log or info you need.

`git clone https://github.com/gnif/vendor-reset.git
cd vendor-reset
dkms install .
echo "vendor-reset" >> /etc/modules
update-initramfs -u

cat << EOF >>  /etc/systemd/system/vreset.service
[Unit]
Description=AMD GPU reset method to 'device_specific'
After=multi-user.target
[Service]
ExecStart=/usr/bin/bash -c 'echo device_specific > /sys/bus/pci/devices/0000:04:00.0/reset_method'
[Install]
WantedBy=multi-user.target
EOF
systemctl enable vreset.service && systemctl start vreset.service
`

lsmod |grep reset
`vendor_reset          122880  0
`

journalctl -xb | grep reset
`kernel: vendor_reset: module verification failed: signature and/or required key missing - tainting kernel
kernel: vendor_reset_hook: installed
systemd-modules-load[695]: Inserted module 'vendor_reset'
systemd[1]: Started vreset.service - AMD GPU reset method to 'device_specific'.
░░ Subject: A start job for unit vreset.service has finished successfully
░░ A start job for unit vreset.service has finished successfully.
systemd[1]: vreset.service: Main process exited, code=exited, status=1/FAILURE
░░ An ExecStart= process belonging to unit vreset.service has exited.
systemd[1]: vreset.service: Failed with result 'exit-code'.
░░ The unit vreset.service has entered the 'failed' state with result 'exit-code'.
kernel: vfio-pci 0000:04:00.0: Unsupported reset method 'device_specific'
`

ballerburg9005, 2024-12-06:
Did you try adding your product ID to one of the four possible reset strategies in device-db.h ?

This PR improves the README, so you understand how to do it: https://github.com/gnif/vendor-reset/pull/89

DocMAX, 2024-12-06:
yes i did

ballerburg9005, 2024-12-06:
Which one did you pick?

I got it working with 5600G, which is also Cezanne but I believe it is Vega 7.

ballerburg9005, 2024-12-06:
I believe from the log you added the wrong product ID or there is something odd with your kernel, because it fails switching reset_method to device_specific, which I think can be done for almost all PCIe devices, except for root ports maybe and such stuff.

kichappa, 2025-05-13:
I was able to solve the `Unsupported reset method 'device_specific'` error by making sure vendor-reset is integrated with initramfs. 

[According to @gnif ](https://forum.level1techs.com/t/amd-polaris-vega-navi-reset-project-vendor-reset/163801/71), this is system dependent, but here are my steps anyway:

1. Add `vendor-reset` into `/etc/initramfs-tools/modules`.
    ```
    # echo 'vendor-reset' >> /etc/initramfs-tools/modules
    # update-initramfs -u -k all
    ```
2. After which, verify:
    ```
    # lsinitramfs /boot/initrd.img-$(uname -r) | grep vendor
    usr/lib/modules/6.8.12-10-pve/updates/dkms/vendor-reset.ko
    ```

Gituser1010, 2025-08-15:
@ballerburg9005 I’m in the same boat — tried every reset method for my iGPU (Vega 7 Mobile) on an AMD Ryzen 5625U, but no luck so far. Has anyone actually managed to get Vendor Reset working on mobile GPUs?

closed 2025-08-17 by david-ng-hk

david-ng-hk, 2025-08-17:
Hi @BSD-gu yes, after passthrough the iGPU to the VM, 
and inside the VM, I can do windows reboot as many times as I can.
I can also reboot the VM in pve host as many times I want to.

before this vendor reset, after each window reboot, I have to reboot the pve host.
so this reset really works.
but in 5825u, you habe to used the vbios which I attached in my previos post to make it work.

BSD-gu, 2025-08-17:
Could you please share what reset strategy you’re using with vendor-reset? I have slightly different hardware (AMD Ryzen 5625U with Vega 7), and also wondering which version of Proxmox you’re on. Thanks!

gabimem, 2025-08-31:
The only thing that's resolved here is getting the passthrough to work, but not the reset. The vBIOS is the solution for the passthrough. If the reset somehow worked for you, it's because you didn't have that problem on your hardware, since the standard version of vendor-reset doesn't support that specific hardware.

I've tried every possible option, even the most up-to-date fork (https://github.com/lowell80/vendor-reset/), and I haven't been able to get it to work. What @david-ng-hk mentions is quite inconsistent and incomplete. I appreciate his contribution, but I think he's confusing other hardware he's using with the specific one in this issue.

The attached vBIOS, while it also works for me for the passthrough, I see that it refers to Cezanne inside, while the one I have specifically for the 15e7 is Barcelo (and that's what's documented in the Linux kernel).

It would be interesting to have a better explanation of the procedure you followed to get it working, and if it's only on Windows, or if you have something installed on Windows for that.

david-ng-hk, 2025-08-31:
Hi @gabimem thank you for reminding me, 
In my windows 10 pro, I did install this
https://github.com/inga-lovinde/RadeonResetBugFix
this is the key to make the reset work, please try
just get the above source code, and compile it, I also attached a version that was compiled by me, I scan with Avast.

[RadeonResetBugFixService pw 12345678.zip](https://github.com/user-attachments/files/22062549/RadeonResetBugFixService.pw.12345678.zip)

Hi @BSD-gu please try above.
the reason why 0000:03:00.0 change to 0000:05:00.0 is, when sometimes you add a new hardware, the original graphic card id will change.

Nightdavisao, 2025-09-15:
Has anyone successfully passed through the 5700U? I literally tried everything I could, including installing the vendor-reset fork. vendor-reset runs its routine to reset the card, but the Windows guest’s screen gets corrupted (weirdly enough, only the Windows cursor is fine; I can see it above the garbled output).