gnif/vendor-reset #78: Build Error: strlcpy removed in kernel 6.8.0-rc1

GaM3r2Xtreme, 2024-03-25:
Current arch package "linux 6.8.1.arch1-1" seems to break when dkms hook attempts to rebuild vendor-reset from AUR package "vendor-reset-dkms-git r114.4b466e9-1". There weren't any build issues experienced with a previous version of the arch "linux 6.7.9.arch1-1" package.

After digging into make log files and searching the web, it looks like the kernel function 'strlcpy' has been removed and replaced by 'strscpy'. I was able to compile vendor-reset after replacing the function call in atom.c and running "dkms install --no-depmod vendor-reset/r114.4b466e9 -k 6.8.1-arch1-1".

Make log file dump, and a tracing page about the kernel change below:
https://pastebin.com/XUCt98Fm
https://review.lttng.org/c/lttng-modules/+/11701

RayOfLight1, 2024-04-11:
I'm on gentoo (6.8.5) and your replacement of an L with an S let me build it too, thank you :)

Slawka, 2024-05-16:
proxmox
6.8.4-3-pve

replace

`dkms install .`

```root@pm:~/vendor-reset# dkms install .
Sign command: /lib/modules/6.8.4-3-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/vendor-reset/0.1.1/source -> /usr/src/vendor-reset-0.1.1

Building module:
Cleaning build area...
make -j16 KERNELRELEASE=6.8.4-3-pve KDIR=/lib/modules/6.8.4-3-pve/build....
Signing module /var/lib/dkms/vendor-reset/0.1.1/build/vendor-reset.ko
Cleaning build area...

vendor-reset.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.4-3-pve/updates/dkms/
depmod....
```