gnif/vendor-reset #82: Is it possible to make this work for the 780M on an 7840hs?

10bn, 2024-08-16:
Is it possible to make this work for the 780M on an 7840hs?

Raltar, 2024-08-23:
I have a system like this too and would be willing to help but would need guidance on where to start.  I don't think it's as simple as just adding Pheonix to the list.

ballerburg9005, 2024-12-06:
I was also deceived by first impressions, that it wouldn't be as simple as adding the product ID to the list, for the iGPU of Ryzen 5 5600G (Cezanne / Vega 7 / CHIP_RENOIR).

However it was as simple as adding the product ID to the list, and simply deploying trial and error on one of the four reset methods of vendor-reset can save you a lot of time, versus actually understanding the solution.

Check out my PR to improve the README, for more information:

https://github.com/gnif/vendor-reset/pull/89

smarticz, 2024-12-20:
@up

kichappa, 2025-03-07:
@ballerburg9005 which one worked for you among the four?

ballerburg9005, 2025-03-13:
_AMD_NAVI14(op) for iGPU 5600G

lowell80, 2025-04-13:
After being inspired by @ballerburg9005's comments and PR, I messed around adding support for my 7840HS w/ Radeon 780M mini PC and got it working!  (This was before I saw your branch, @10bn; which I tested and it seems to work too, but the kernel messages seem to go way, and I can't quite tell what all is going on with some of the changes you've made.  Although a scripted approach at trying all the different reset approaches looked clever, but I didn't stare long enough to figure out how to use it)

I dug around in the network of forks for a bit; and noticed that half of the forks are just merging some combination of the same 3 (open) PRs; and the other half are people adding support for just a single new device id.  

I've taken the liberty of just merging a bunch of them that seemed reasonable into (yet-another-fork) in the hopes that it will save others some time. I also update the README to try to better represent what's worked for others.

Oh, and thanks @ballerburg9005 for actually noting that you have to copy the udev rules manually.  I haven't messed with dkms in many years, but I assumed that either it or the Makefile would certainly have installed the udev rules, or the README or one of blog post, ...; but NOPE it was just as missing step that never got listed.

For anyone interested, my consolidation effort lives here:  https://github.com/lowell80/vendor-reset/

kichappa, 2025-04-13:
@lowell80 thank you for your work. Am I understanding correctly that you were able to fix the reset bug after it appeared?

lowell80, 2025-04-13:
@kichappa, short answer: yes

I have a Mini PCs with 7840hs/780M combo that's part of a homlab proxmox cluster that runs under my desk, that I wanted to dedicate as a Ubuntu desktop while also being able to pick up some slack in the cluster when the other nodes need to restarted or whatever.  So it's been a several month long journey, when my time/priorities/focus/technical-know-how finally aligned to get GPU passthrough to work at all.   That was last week, and I expected that I may have device reset issues based on what I read; and it was easy to confirmed by stopping/starting the VM

The changes needed for just he 780M is just this one commit: https://github.com/lowell80/vendor-reset/commit/9ffeb25f075ef86229444cebbaa83766794b602e

---

## Steps
Hopefully this helps someone, here are the steps I've taken to make this work:

My system:
```bash
root@proxmox-node01:~# lspci -vnn | grep VGA
c7:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Phoenix1 [1002:15bf] (rev c7) (prog-if 00 [VGA controller])

```

Add kernel parameters to `/etc/kernel/cmdline`
`amd_iommu=on iommu=pt`  (Not 100% sure this is necessary, but it's a step I took)

Install vendor-reset:  (using my branch here)
```bash
git clone
https://github.com/lowell80/vendor-reset
cd vendor-reset
dkms install .
cp udev/99-vendor-reset.rules /etc/udev/rules.d/
```

I added the following to `/etc/modules`:
```
vfio
vfio_iommu_type1
vfio_pci
vfio_virqfd

vendor-reset
```

```bash
update-initramfs -k all -u
proxmox-boot-tool refresh
# vs GRUB; required on proxmox using ZFS

echo "blacklist amdgpu" >> /etc/modprobe.d/blacklist.conf
echo "blacklist radeon" >> /etc/modprobe.d/blacklist.conf
```

Download and copy ROM file, and tell proxmox to use it:  (Without this guest kernel reports:  _amdgpu: Unable to locate a BIOS ROM_; all attempts at extracting the ROM image myself failed miserably).  Thanks @isc30!
```bash
wget https://github.com/isc30/ryzen-gpu-passthrough-proxmox/raw/refs/heads/main/vbios_7840hs.bin

cp vbios_7840hs.bin /usr/share/kvm/
```

Add to proxmox VM config:
```
hostpci0: mapping=minipc-amdgpu,x-vga=1,pcie=1,romfile=vbios_7840hs.bin
```

Double check that kernel cmdline and modules took effect:

```
root@proxmox-node01:~# cat /proc/cmdline 
initrd=\EFI\proxmox\6.8.12-9-pve\initrd.img-6.8.12-9-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs amd_iommu=on iommu=pt
```

Once VM starts, confirm that the 'reset_method' has been set successfully (by the udev rule):
```bash
[... 83 lines cut ...]
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: performing reset
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: bus reset disabled? yes
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? no
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: begin psp mode 1 reset
[Sun Apr 13 00:44:30 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: PSP did not acknowledger reset
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: failed to reset device
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: performing post-reset
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: reset result = 0
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: version 1.1
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: performing pre-reset
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: performing reset
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: bus reset disabled? yes
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? no
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: Clearing scratch regs 6 and 7
[Sun Apr 13 00:44:31 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: begin psp mode 1 reset
[Sun Apr 13 00:44:32 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: timed out waiting for PSP to reach valid state, but continuing anyway
[Sun Apr 13 00:44:32 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: PSP did not acknowledger reset
[Sun Apr 13 00:44:32 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: failed to reset device
[Sun Apr 13 00:44:32 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: performing post-reset
[Sun Apr 13 00:44:32 2025] vfio-pci 0000:c7:00.0: AMD_NAVI10: reset result = 0
```

ballerburg9005, 2025-04-13:
I don't want to spam this issue, but I noticed that my system would crash every few days when I had the vfio_pci and vendor-reset stuff loading at boot, without actually using vendor-reset. I am not sure anymore, but I think the first thing I tried was just to blacklist vendor-reset module, and when this didn't cut it I unloaded all the vfio_pci stuff and changed cmd line also.

So beware if this is happening to your system as well.

r00teniy, 2025-05-03:
@lowell80 Hello, i tried your version on my 7840hs/780M and while i can pass through GPU normally it wont reset, because it can't locate BIOS ROM. Your  fork doesn't have issues, so i'm writing here. Maybe you know what could result in this error. I properly have vbios romfile (tried both dowloaded and extracted)
Here is the error:
`May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: version 1.1
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: performing pre-reset
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: performing reset
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
May 03 21:00:10 pve2 kernel: vendor-reset-drm: Unable to locate a BIOS ROM
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: amdgpu_get_bios failed: 0
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: failed to reset device
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: performing post-reset
May 03 21:00:10 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI10: reset result = 0`

r00teniy, 2025-05-04:
> And maybe that's just a PVE thing? idk. So if your not using PVE, I'm sure there's some other way to specify the whole `romfile=vbios_7840hs.bin` bits.

I do use Proxmox and only difference in our setups is grub vs proxmox-boot-tool, because i don't use ZFS on my. I do have romfile specified (won't work without it) and it named same way. Will try to look through code to find part related to this error then, I am not familiar with C sadly.

kichappa, 2025-05-13:
@lowell80 I tried your repo but I keep getting 

```
# dmesg -T | grep 75:00
...
pci 0000:75:00.0: Unsupported reset method 'device_specific'
...
```

General Info:
```
# lspci | grep Rembrandt
75:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [Radeon 680M] [1002:1681] (rev c9)
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [Radeon 680M] [1002:0124]
        Kernel driver in use: vfio-pci
        Kernel modules: amdgpu
75:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt Radeon High Definition Audio Controller [1002:1640]
        Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt Radeon High Definition Audio Controller [1002:1640]
        Kernel driver in use: vfio-pci
        Kernel modules: snd_hda_intel
```

These are the steps I'm following:

1. `git clone https://github.com/lowell80/vendor-reset`,
2. `cd vendor-reset`,
3. Remove earlier `vendor-reset`, `dkms remove -m vendor-reset -v 0.1.1 --all`
4. Clean earlier make, `make clean`,
5. Add these lines:
To `device-db.h`
```
#define _AMD_NAVI10(op) \
    ...
    {PCI_VENDOR_ID_ATI, 0x1681, op, DEVICE_INFO(AMD_NAVI10)}, \ 
```
and to `/etc/udev/rules.d/99-vendor-reset.rules`
```
ACTION=="add", SUBSYSTEM=="pci", ATTR{vendor}=="0x1002", ATTR{device}=="0x1681", RUN+="/bin/sh -c '/sbin/modprobe vendor-reset; echo device_specific > /sys$env{DEVPATH}/reset_method'"
```
6. `make`,
7. `dkms install .`,
8. `update-initramfs -u -k all`
9. `reboot`

When I check `reset_method`
```
# cat /sys/bus/pci/devices/0000:75:00.0/reset_method
device_specific
```

But I still get `pci 0000:75:00.0: Unsupported reset method 'device_specific'`. Thoughts?

kichappa, 2025-05-13:
@r00teniy  Thanks for responding! I feel so lost already. 

I actually have the line in the cloned repo's file, so the line is persistent across installs. Also, the `# cat /sys/bus/pci/devices/0000:75:00.0/reset_method` command confirms that `device_specific` is put in. The problem (I think) is that `device_specific` isn't working.

r00teniy, 2025-05-13:
@kichappa this exact error happens if vendor reset is either not loaded via modules or doesn't have your device in rules, system doesn't know what to do, because nothing responds to device_specific trigger for this device.
Can you try `modprobe vendor-reset ` , check rules in /etc and `udevadm control --reload-rules && udevadm trigger`  before starting VM to see if it would change anything?

kichappa, 2025-05-13:
@r00teniy Can you give me a sanity check first?

I have a windows VM that has the passthrough set up. The first time I setup the passthrough, the iGPU was accessible by the VM. However, the reset bug hit the minute the host rebooted.

I am trying to get this repo to solve this issue. Should I be performing these steps (that I mentioned I did) with the `hostpci0: 0000:75:00.0,pcie=1,romfile=vbios_6600H.bin` line still present in the VM config file (the VM is not starting up, so it isn't running)?

kichappa, 2025-05-13:
```# modprobe vendor-reset
   # (no output)
```

Running `udevadm control --reload-rules && udevadm trigger`
```
()
101 is starting, disabling d3cold for selected PCI device.
0
TASK ERROR: start failed: QEMU exited with code 1
```
Another thing I notice is that the last time I tried fixing this, around the time my first comment in this issue was made, starting VM gave me an output that specifically mentioned `ioctl` (I lost the log now).

r00teniy, 2025-05-13:
@kichappa  after you got reset bug you need to restart Proxmox, because reset bug can only be prevented, not fixed after.
Adding vendor-reset to modules should fix the problem, it wasn't loaded, so it didn't work.
Try to restart Poxmox and run/stop/run VM again. you should get new messages in log from vendor reset (you can look my message from last week with log output).

kichappa, 2025-05-13:
> [@lowell80](https://github.com/lowell80) Hello, i tried your version on my 7840hs/780M and while i can pass through GPU normally it wont reset, because it can't locate BIOS ROM. Your fork doesn't [...]

@r00teniy just found out that this happens in a method that doesn't work. When I swapped methods (NAVI10 -> POLARIS10), the `Unable to locate a BIOS ROM` error disappeared.

kichappa, 2025-05-13:
> _AMD_NAVI14(op) for iGPU Ryzen 5 5600G
> 
> This decision was based on the AMD comments in the source code, since it is a GPU integrated into the CPU.

@ballerburg9005 can you also post the successful reset output in journalctl, `journalctl -b 0 | grep reset?

r00teniy, 2025-05-13:
> ```
> 
> ```
> 
> 
>     
>   
> 
> [@r00teniy](https://github.com/r00teniy) just found out that this happens in a method that doesn't work. When I swapped methods (NAVI10 -> POLARIS10), the `Unable to locate a BIOS ROM` error disappeared.

Thank you, will try POLARIS10 and other methods on my 780m

kichappa, 2025-05-13:
Note for future readers:

Errors from each were different. Here is the catalog of errors I got:

- `ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment`
- `AMD_POLARIS12: CLOCK_CNTL: 0x1, PC: 0xffffffff`
- `vendor-reset-drm: Unable to locate a BIOS ROM`
- `AMD_NAVI14: failed to reset device`

r00teniy, 2025-05-13:
@kichappa did any of the methods work for you?

kichappa, 2025-05-13:
@r00teniy I think maybe VEGA20 worked for me, but my VM won't start at all. My idea is to reinstall the VM and try to reproduce the bug again.

VEGA20 output:
```
May 13 04:19:06 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA20: version 1.0
May 13 04:19:06 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA20: performing pre-reset
May 13 04:19:06 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA20: performing reset
May 13 04:19:06 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA20: psp mode1 reset succeeded
May 13 04:19:06 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA20: performing post-reset
May 13 04:19:06 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA20: reset result = 0
```
VEGA10 output:
```
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: version 1.0
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: performing pre-reset
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: performing reset
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: amdgpu_get_bios failed: 0
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: failed to reset device
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: performing post-reset
May 13 04:23:43 pve kernel: vfio-pci 0000:75:00.0: AMD_VEGA10: reset result = 0
```
NAVI14 output:
```
[Tue May 13 04:29:16 2025] vfio-pci 0000:75:00.0: AMD_NAVI14: performing pre-reset
[Tue May 13 04:29:16 2025] vfio-pci 0000:75:00.0: AMD_NAVI14: performing reset
[Tue May 13 04:29:16 2025] vendor-reset-drm: Unable to locate a BIOS ROM
[Tue May 13 04:29:16 2025] vfio-pci 0000:75:00.0: AMD_NAVI14: failed to reset device
[Tue May 13 04:29:16 2025] vfio-pci 0000:75:00.0: AMD_NAVI14: performing post-reset
[Tue May 13 04:29:16 2025] vfio-pci 0000:75:00.0: AMD_NAVI14: reset result = 0
```
NAVI10 output:
```
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: version 1.1
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: performing pre-reset
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: performing reset
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: amdgpu_get_bios failed: 0
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: failed to reset device
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: performing post-reset
May 13 04:15:35 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI10: reset result = 0
```
NAVI12 output:
```
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: version 1.1
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: performing pre-reset
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: performing reset
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: amdgpu_get_bios failed: 0
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: failed to reset device
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: performing post-reset
May 13 04:38:08 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI12: reset result = 0
```
NAVI23 output:
```
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: version 1.1
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: performing pre-reset
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: performing reset
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: amdgpu_get_bios failed: 0
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: failed to reset device
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: performing post-reset
May 13 04:45:52 pve kernel: vfio-pci 0000:75:00.0: AMD_NAVI23: reset result = 0
```
All POLARIS outputs are the same:
```
May 13 04:05:28 pve kernel: vfio-pci 0000:75:00.0: AMD_POLARIS10: version 1.1
May 13 04:05:29 pve kernel: vfio-pci 0000:75:00.0: AMD_POLARIS10: performing pre-reset
May 13 04:05:29 pve kernel: vfio-pci 0000:75:00.0: AMD_POLARIS10: performing reset
May 13 04:05:29 pve kernel: vfio-pci 0000:75:00.0: AMD_POLARIS10: CLOCK_CNTL: 0x1, PC: 0xffffffff
May 13 04:05:29 pve kernel: vfio-pci 0000:75:00.0: AMD_POLARIS10: performing post-reset
May 13 04:05:29 pve kernel: vfio-pci 0000:75:00.0: AMD_POLARIS10: reset result = 0
```

r00teniy, 2025-05-13:
@kichappa  i do believe reset result = 0 means it failed, but i don't have PC with working setup to confirm it.

kichappa, 2025-05-13:
@r00teniy I believe I have identified some issue with the vbios. When I run 

```
# rom-parser vbios_6600H.bin
    Valid ROM signature found @0h, PCIR offset 1bch
    PCIR: type 0 (x86 PC-AT), vendor: 1002, device: 1681, class: 030000
    PCIR: revision 0, vendor revision: 2011
    Last image
```

The vbios I extracted from @isc30's [ryzen-proxmox repo](https://github.com/isc30/ryzen-gpu-passthrough-proxmox) has an offset of `0x1bc` but the PCI firmware spec mandates PCI expansion ROMs start at offsets divisible by 512b (`0x200`).

kichappa, 2025-05-14:
> No, I am too lazy. But I can tell you that it looked like nothing special. It just went through the reset stuff like with the card it was written for.

Cool. I'm more surprised on how for some iGPUs, `amdgpu_read_platform_bios()` loads the bios successfully while on others, it fails. I went through the code to see all the logs that should appear on successfully reset, and looks like my output has all of them.

r00teniy, 2025-05-14:
for me all NAVI options give same error with vbios patch:
`
May 14 21:37:33 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: version 1.1
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: performing pre-reset
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: performing reset
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
May 14 21:37:35 pve2 kernel: ATOM BIOS: 113-PHXGENERIC-001
May 14 21:37:35 pve2 kernel: vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: bus reset disabled? yes
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? no
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: Clearing scratch regs 6 and 7
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: begin psp mode 1 reset
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: timed out waiting for PSP to reach valid state, but continuing anyway
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: PSP did not acknowledger reset
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: failed to reset device
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: performing post-reset
May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: reset result = 0
`
same as for you vega20 doesn't give an error, but still doesn't work:
`May 14 21:43:27 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: version 1.0
May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: performing pre-reset
May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: performing reset
May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: psp mode1 reset succeeded
May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: performing post-reset
May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: reset result = 0`
I guess we are still missing something

kichappa, 2025-05-14:
> for me all NAVI options give same error with vbios patch: 
> ```
> May 14 21:37:33 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: version 1.1 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: performing pre-reset 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: performing reset 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment 
> May 14 21:37:35 pve2 kernel: ATOM BIOS: 113-PHXGENERIC-001 
> May 14 21:37:35 pve2 kernel: vendor-reset-drm: atomfirmware: bios_scratch_reg_offset initialized to 4c 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: bus reset disabled? yes 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: SMU response reg: 0, sol reg: 0, mp1 intr enabled? no, bl ready? no 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: Clearing scratch regs 6 and 7 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: begin psp mode 1 reset 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: timed out waiting for PSP to reach valid state, but continuing anyway 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: PSP did not acknowledge reset 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: failed to reset device 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: performing post-reset 
> May 14 21:37:35 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_NAVI12: reset result = 0
> ``` 
> same as for you vega20 doesn't give an error, but still doesn't work: 
> ```
> May 14 21:43:27 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: version 1.0 
> May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: performing pre-reset 
> May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: performing reset 
> May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: psp mode1 reset succeeded 
> May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: performing post-reset 
> May 14 21:43:28 pve2 kernel: vfio-pci 0000:c8:00.0: AMD_VEGA20: reset result = 0
> ``` 
> I guess we are still missing something

Looks like your `psp_wait_for` isn't seeing a change in the register. I don't know if this is a hardware issue, but I would just try waiting longer:
```
int psp_wait_for(struct amd_fake_dev *adev, uint32_t reg_index,
                 uint32_t reg_val, uint32_t mask, bool check_changed)
{
  uint32_t val;
  int i;

  for (i = 0; i < 100000; i++) // <--------- Change 100000 to 1000000
  {
    val = RREG32(reg_index);
    if (check_changed)
    {
      if (val != reg_val)
        return 0;
    }
    else
    {
      if ((val & mask) == reg_val)
        return 0;
    }
    udelay(1);
  }

  return -ETIME;
}
```

r00teniy, 2025-05-14:
@kichappa adding 0 to value didn't change much, same error. Probably related to hardware.
I decided to do something stupid and got strange result:
instead of passing .0 and .1 separately i passed all functions of device with vega20 reset method.
Windows actually starts after you shut it down, but GPU is disabled due to error, you can enable it, but other software won't see it after enabling

kichappa, 2025-05-14:
Oh, I'm sorry NAVI isn't working for you. Can you elaborate on this?
> instead of passing .0 and .1 separately i passed all functions of device with vega20 reset method.

It so happened that I pushed an ambitious `amdgpu_bios.c` patch that auto extracts the vbios and my proxmox wouldn't start at all. So, I have now reinstalled proxmox without any VM. I've to make everything from scratch again.

r00teniy, 2025-05-14:
> Oh, I'm sorry NAVI isn't working for you. Can you elaborate on this?
@kichappa 
So i have 0000:c8:00.0 GPU and 0000:c8:00.1 audio, normally i only pass them.
But instead i checked pass all functions checkbox and it passed full 0000:c8:00, i added romfile to it and tried. Results:
1) first time it starts normally, but you get rest of devices with generic drivers (probably because they weren't blacklisted)
2) you can reboot into windows, but if you go to device manager GPU has an error, you can restart it, but other software still won't see i
3) it was on vega20, but maybe it will work for you on navi, because you don't get errors

kichappa, 2025-05-15:
@r00teniy I see. I ended up reinstalling the VM, and RadeonResetBugFix seems to perform things quite well this time around. How did you extract the VBIOS? Did you use the UBU tool (and BIOS downloaded from your oem) or through the vbios.c file from isc30's repo? I found the former to be problematic.

Also, I came across [this](https://github.com/inga-lovinde/RadeonResetBugFix/issues/6#issuecomment-2426475924). Maybe this will help you, especially if your device is RDNA3.

r00teniy, 2025-05-15:
@kichappa did you use just RadeonResetBugFix or together with vendor-reset?
I tried both extract with vbios.c and rom provided in isc30 repo. 
Will try UBU tool and try oznakn's suggestion. Thank you.

kichappa, 2025-05-15:
>  @kichappa did you use just RadeonResetBugFix or together with vendor-reset?
> I tried both extract with vbios.c and rom provided in isc30 repo. 
> Will try UBU tool (i am using seabios and i don't think aoostar provides BIOS roms, but will check) and try oznakn's suggestion. Thank you.

Only RadeonResetBugFix, no vendor reset since I found two new errors when I use vendor reset (despite it saying reset successful for Vega and Navi). [Here's](https://github.com/isc30/ryzen-gpu-passthrough-proxmox/discussions/18#discussioncomment-8627679) a good UBU tool tutorial.

I can confirm now that the VM pass through is successful and I can shut down and restart the VM without any issues.

xiongyw, 2025-07-01:
> When I check `reset_method`
> 
> ```
> # cat /sys/bus/pci/devices/0000:75:00.0/reset_method
> device_specific
> ```
> 
> But I still get `pci 0000:75:00.0: Unsupported reset method 'device_specific'`. Thoughts?

I encountered the similar issue today, `dmesg` reporting `vfio-pci 0000:c5:00.0: Unsupported reset method 'device_specific'`. It turns out the the `vendor_reset` module database does not contain my device id (1002:1900) yet. After recompile/install the module, the problem resolved. This suggests that when the `reset_method` is set to `device_specific`, the `vendor_reset` module is consulted somehow.

xiongyw, 2025-07-02:
> ```
> > vendor-reset-drm: Unable to locate a BIOS ROM 
> ```
> 
> [@xiongyw](https://github.com/xiongyw) how did you generate the bios?

I generate the bios using the `vbios.c`.

Btw, today it seems that I found a "perfect" (?) workaround for my scenario, no vendor_reset, no AMDResetBugFix. But using both `devcon` in VM and pci `remove/rescan` on host combined. The idea comes from [this post](https://forum.level1techs.com/t/navi-reset-kernel-patch/147547/83). Btw, no "suspend to ram" is needed for my scenario.

The `devcon` install/setup procedure I am following is [this one](https://forum.level1techs.com/t/linux-host-windows-guest-gpu-passthrough-reinitialization-fix/121097). I use [devcon installer](https://github.com/Drawbackz/DevCon-Installer) to install devcon following the instruction [here](https://superuser.com/a/1751153) for my win11 pro.

The following is my hook script for windows vm 100:

```
# cat /var/lib/vz/snippets/vm100-hooks.sh
#!/usr/bin/bash

# ref: https://gist.github.com/kiler129/215e2c8de853209ca429ad5ed40ce128

set -e -o errexit -o pipefail -o nounset

# Do not modify these variables (set by Proxmox when calling the script)
vmId="$1"
runPhase="$2"
echo ">>> Entering $runPhase on VM=$vmId" > /dev/kmsg

case "$runPhase" in
    pre-start)
      echo "### rescan pcie bus..." > /dev/kmsg
      echo 1 > /sys/bus/pci/rescan
      sleep 2
      ;;
    post-start)
      ;;
    pre-stop)
      echo "### remove gpu/snd devices..." > /dev/kmsg
      GFX_PATH=$(dirname $(grep -l "PCI_ID=1002:1900" /sys/bus/pci/devices/*/uevent))
      SND_PATH=$(dirname $(grep -l "PCI_ID=1002:1640" /sys/bus/pci/devices/*/uevent))
      echo 1 > ${GFX_PATH}/remove
      sleep 1
      echo 1 > ${SND_PATH}/remove
      sleep 1
      ;;
    post-stop)
      echo "### rescan pcie bus..." > /dev/kmsg
      echo 1 > /sys/bus/pci/rescan
      sleep 2
      ;;
    *)
      echo "### Unknown run phase \"$runPhase\"!" > /dev/kmsg
      ;;
esac

echo "<<< Leaving $runPhase on VM=$vmId" > /dev/kmsg
```

Now from windows VM I can Restart/Shutdown gracefully, also from pve host i can qm `start/stop` gracefully. The nice thing turns out that this two methods are somewhat *not* overlapping, meaning that Restart/Shutdown from VM does not trigger the VM hook. 

So far I am content with this workaround. Will experience more to see how it goes.

Hawxxer, 2025-08-03:
I guess but i am not sure I have similar issues with an Minisforum UM870M - 8745H - 780M.
There are 2 different behaviours depending on the client VM with Proxmox 6.8.12-13-pve (Opt-In Kernel 6.14.8-2-bpo13-pve gives me multiply other issues so i skip that for now):
Ubuntu (24.04) (UEFI, Q35,X-VGA=1,Display:None(Setting) - VBIOS and AMDGOPDriver for 8745HS close enough)
(Passthrough 5 Devices: GPU, AUDIO, Encryption, Audio Coprocessor, Audio Device (xx:00.0, xx:00.1,xx:00.2, xx:00.5, xx:00.6 - Also all bound directly to vfio-pci and boot -> This prevents random crashes when starting / stoping VMs seen in dmesg as kernel pointer issues) : 
Normal start/shutdown works fine multiple times, no reset bug 
Force Stop VM -> No Recovery, I can restart the vm, which runs but the video output is stuck to last image when VM was stopped forcefully.

Win11 (UEFI, Q35,X-VGA=1, Display:None(Setting) - VBIOS and AMDGOPDriver for 8745HS close enough) X-VGA=1
Same as above all 5 Devices
After Driver install the VM boots up 1 time with Video output, after that it wont give any video output -> Hard reset of host required (also i can't shutdown the host, it's responsive -> I can send the command poweroff, but it wont fully shut down) 
Using radeonResetBugFix works if the VM is shut down correctly. 

I tried all four possibilities:
Polaris10 -> Bios found but not fixing issues after forcefull VM Stop (Ubuntu)
VEGA10 -> Bios NOT found and not fixing issues after forcefull VM Stop (Ubuntu)
Vega20 -> Bios found but not fixing issues after forcefull VM Stop (Ubuntu)
NAVI10 -> Bios NOT found and not fixing issues after forcefull VM Stop (Ubuntu)
(All other options in device-db.h are one of this 4 possibilities)

In all 4 possibilities I get:
`
[    0.858746] pci 0000:c4:00.0: Unsupported reset method 'device_specific'
[    3.071172] vendor_reset: module verification failed: signature and/or required key missing - tainting kernel
[    3.101300] vendor_reset_hook: installed
`
**In the beginning, I am not sure if this is normal, maybe someone can tell** -> Could this be an issue of the module loading sequence? Its loaded before vfio:
`
# /etc/modules: kernel modules to load at boot time.
vendor-reset
vfio
vfio_iommu_type1
vfio_pci
`
But of course I have configurations in my modprobe, that I guess tell the system to load vfio-pci early:
`
cat /etc/modprobe.d/vfio.conf 
options vfio-pci ids=1002:1900,1002:1640,1022:15c7,1022:15e2,1022:15e3
softdep radeon pre: vfio-pci
softdep amdgpu pre: vfio-pci
softdep snd_hda_intel pre: vfio-pci
softdep snd_pci_ps pre: vfio-pci
softdep ccp pre: vfio-pci
`

I will try fixing the the static bios search and I will extract my own vbios and will report back if this helps. But a solution on host side would be the best overall fix, instead of relying on the client vm

Nazgile94, 2025-08-05:
`ug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: enabling device (0400 -> 0403)
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: version 1.1
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing pre-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
Aug 05 23:35:37 Proxmox-VE kernel: vendor-reset-drm: Unable to locate a BIOS ROM
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: amdgpu_get_bios failed: 0
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: failed to reset device
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing post-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: reset result = 0
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: version 1.1
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing pre-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
Aug 05 23:35:37 Proxmox-VE kernel: vendor-reset-drm: Unable to locate a BIOS ROM
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: amdgpu_get_bios failed: 0
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: failed to reset device
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing post-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: reset result = 0`

Nazgile94, 2025-08-05:
`ug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: enabling device (0400 -> 0403)
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: version 1.1
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing pre-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
Aug 05 23:35:37 Proxmox-VE kernel: vendor-reset-drm: Unable to locate a BIOS ROM
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: amdgpu_get_bios failed: 0
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: failed to reset device
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing post-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: reset result = 0
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: version 1.1
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing pre-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: ROM [??? 0x00000000 flags 0x20000000]: can't assign; bogus alignment
Aug 05 23:35:37 Proxmox-VE kernel: vendor-reset-drm: Unable to locate a BIOS ROM
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: amdgpu_get_bios failed: 0
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: failed to reset device
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: performing post-reset
Aug 05 23:35:37 Proxmox-VE kernel: vfio-pci 0000:0d:00.0: AMD_NAVI10: reset result = 0`

getting same , bios not found
Ryzen 8700GE - 780M