gnif/vendor-reset #88: DKMS Install Issue on 6.12.1

luyu-wu, 2024-11-26:
Getting a new error after updating my kernel on Arch Linux.

Sign command: /usr/lib/modules/6.12.1-2-cachyos/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/vendor-reset/0.1.1/source -> /usr/src/vendor-reset-0.1.1

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j16 KERNELRELEASE=6.12.1-2-cachyos KDIR=/usr/lib/modules/6.12.1-2-cachyos/build LLVM=1
Error! Bad return status for module build on kernel: 6.12.1-2-cachyos (x86_64)
Consult /var/lib/dkms/vendor-reset/0.1.1/build/make.log for more information.

The error log can be seen here:
<details>
<summary>DKMS Make Log</summary>
`DKMS make.log for vendor-reset-0.1.1 for kernel 6.12.1-2-cachyos (x86_64)
Mon Nov 25 09:35:03 PM EST 2024
make -C /usr/lib/modules/6.12.1-2-cachyos/build M=/var/lib/dkms/vendor-reset/0.1.1/build modules
make[1]: Entering directory '/usr/lib/modules/6.12.1-2-cachyos/build'
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/module.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/vendor-reset-dev.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/hook.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/common.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/compat.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/firmware.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/navi10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/polaris10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/vega10.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/vega20.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_device.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_discovery.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_bios.o
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_atomfirmware.o
/var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.c:91:5: warning: no previous prototype for function 'vendor_reset_ioctl_init' [-Wmissing-prototypes]
   91 | int vendor_reset_ioctl_init(void)
      |     ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.c:91:1: note: declare 'static' if the function is not intended to be used outside of this translation unit
   91 | int vendor_reset_ioctl_init(void)
      | ^
      | static 
/var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.c:96:6: warning: no previous prototype for function 'vendor_reset_ioctl_exit' [-Wmissing-prototypes]
   96 | void vendor_reset_ioctl_exit(void)
      |      ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/ioctl.c:96:1: note: declare 'static' if the function is not intended to be used outside of this translation unit
   96 | void vendor_reset_ioctl_exit(void)
      | ^
      | static 
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/atom.o
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:66:5: warning: no previous prototype for function 'fh_install_hook' [-Wmissing-prototypes]
   66 | int fh_install_hook(struct ftrace_hook *hook)
      |     ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:66:1: note: declare 'static' if the function is not intended to be used outside of this translation unit
   66 | int fh_install_hook(struct ftrace_hook *hook)
      | ^
      | static 
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:96:6: warning: no previous prototype for function 'fh_remove_hook' [-Wmissing-prototypes]
   96 | void fh_remove_hook(struct ftrace_hook *hook)
      |      ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/ftrace.c:96:1: note: declare 'static' if the function is not intended to be used outside of this translation unit
   96 | void fh_remove_hook(struct ftrace_hook *hook)
      | ^
      | static 
2 warnings generated.
  CC [M]  /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/common_baco.o
2 warnings generated.
/var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/amdgpu_device.c:47:6: warning: no previous prototype for function 'amdgpu_device_vram_access' [-Wmissing-prototypes]
[... 55 lines cut ...]
1 error generated.
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/atom.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/vega10_reg_init.c:29:5: warning: no previous prototype for function 'vega10_reg_base_init' [-Wmissing-prototypes]
   29 | int vega10_reg_base_init(struct amd_fake_dev *adev)
      |     ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/vega10_reg_init.c:29:1: note: declare 'static' if the function is not intended to be used outside of this translation unit
   29 | int vega10_reg_base_init(struct amd_fake_dev *adev)
      | ^
      | static 
1 warning generated.
/var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/vega20_reg_init.c:29:5: warning: no previous prototype for function 'vega20_reg_base_init' [-Wmissing-prototypes]
   29 | int vega20_reg_base_init(struct amd_fake_dev *adev)
      |     ^
/var/lib/dkms/vendor-reset/0.1.1/build/src/amd/amdgpu/vega20_reg_init.c:29:1: note: declare 'static' if the function is not intended to be used outside of this translation unit
   29 | int vega20_reg_base_init(struct amd_fake_dev *adev)
      | ^
      | static 
1 warning generated.
make[2]: *** [/usr/lib/modules/6.12.1-2-cachyos/build/Makefile:1946: /var/lib/dkms/vendor-reset/0.1.1/build] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.1-2-cachyos/build'
make: *** [Makefile:8: build] Error 2
`
<\details>

fseoane, 2024-11-27:
Did you checked this?  https://github.com/gnif/vendor-reset/issues/87

luyu-wu, 2024-11-29:
> Did you checked this?  https://github.com/gnif/vendor-reset/issues/87 

Sorry, dunno how I didn't catch that when reviewing issues! Closing now, many thanks.

closed 2024-11-29 by luyu-wu