gnif/vendor-reset #92: will vendor-reset work for Radeon 680M Rembrandt?

wustrong, 2024-12-19:
will vendor-reset work for Radeon 680M Rembrandt on pve 8.3?

lowell80, 2025-04-13:
It's probably possible;  I just solved this for myself on 780M and found that it really it's all that hard to add support for new models.  Really you just need the `lspci -vnn | grep VGA` output, primarily just for the device id primarily, and then try one of 4 different reset methods.  More info in #82.

I just scanned though a ton of forks and consolidated some additional device support.  (I'm not sure 680M/Rembrandt was on the list, but it may be worth checking)

https://github.com/lowell80/vendor-reset/

kichappa, 2025-05-12:
Probably a typo, but for newbies like me, it's confusing!

> and found that it really *isn't* all that hard to add support for new models.

dmuiX, 2025-09-15:
have a look here ;)

https://github.com/gnif/vendor-reset/issues/73