gnif/vendor-reset #94: dkms install does not work on kernel 6.12.7-arch1-1

o-gui-blindao, 2025-01-02:
Not sure if this issue is with the arch kernel or 6.12.7 in general, but when I try to run the `dkms install .` command, it errors, this is the logs that happen when I run the command

`guipro@guipro-PC:~$ cd vendor-reset && sudo dkms install .`

`Sign command: /usr/lib/modules/6.12.7-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/vendor-reset/0.1.1/source -> /usr/src/vendor-reset-0.1.1`

`Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=6.12.7-arch1-1 KDIR=/usr/lib/modules/6.12.7-arch1-1/build`

`Error! Bad return status for module build on kernel: 6.12.7-arch1-1 (x86_64)
Consult /var/lib/dkms/vendor-reset/0.1.1/build/make.log for more information.
guipro@guipro-PC:~/vendor-reset$`

FoxInTango, 2025-01-04:
If the error message in the “make.log” is missing <asm/unaligned.h>,you can change it to <linux/unaligned.h>,then make it or dkms install as you have done.

kutukvpavel, 2025-02-05:
See #86